StefanSchippers/xschem #308: Add metadata to symbols

georgtree, 2025-02-10:
Hello!
is it possible to enhance the format of the symbols (and schematics) with ability to add custom metadata?
The idea is to add an additional information to symbol that is not processing by xschem, but help external tools to extract useful data that could be useful for end users. For me it is a way to unify symbol and additional description, such as: parameters descriptions, element description, etc.
For example:
```
m {{{Name {Default value} Units Description} 
                              {OUTOFFS 0.0 - {Offset of output signal}} 
                              {K 1.0 - {Gain of input signal}} 
                              {OUTLOWLIM -1e12 - {Lower limit of output}} 
                              {OUTUPLIM 1e12 - {Upper limit of output}}
                              {LIMR 1e-6 - {Upper & lower limit smoothing range}}
                              }}
```

So, we can save string there, and then fetch it with
```tcl
xschem getprop instance x3 metadata 
```
maybe such mechanism already exist? thank you

StefanSchippers, 2025-02-10:
symbols and schematics may have metadata at the very beginning, it is actually used to contain license info:

```
v {xschem version=3.4.6 file_version=1.2
*
* This file is part of XSCHEM,
* a schematic capture and Spice/Vhdl/Verilog netlisting tool for circuit
* simulation.
* Copyright (C) 1998-2024 Stefan Frederik Schippers
*
* This program is free software; you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation; either version 2 of the License, or
* (at your option) any later version.
*
* This program is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU General Public License
* along with this program; if not, write to the Free Software
* Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA
}
```

After the first line and before the closing brace anything can be written. 
You only need to escape `{`, `}` and `\` characters.

You can add the text manually or use the xschem command Properties -> Edit License/Header text

you can extract the metadata from the xschem console with the command `xschem get header_text` and written with the command `xschem set header_text "string"`.
If you use xschem commands backslashes to `{`, `}`, `\` will be added automatically (when writing) and removed automatically (when reading). I usualy add needed text manually directly in the file.

Just keep the header line `v {xschem version=3.4.6 file_version=1.2` and put a closing brace `}` after the metadata.

georgtree, 2025-02-10:
Ok got it, so I get header and then remove the `xschem version=3.4.6 file_version=1.2` line right?

StefanSchippers, 2025-02-10:
Consider this symbol file: `amplifier.sym`:

```
v {xschem version=3.4.6 file_version=1.2
author="Stefan Schippers"
version=3.0
description="Verilog A model of a differential amplifier"
}
...
... <rest of symbol xschem data follows
...
```

you can retrieve the data with:

```
% xschem get_tok [xschem get header_text] author
Stefan Schippers
% xschem get_tok [xschem get header_text] description
Verilog A model of a differential amplifier
```

The command xschem get header_text returns the complete metadata:
```
% xschem get header_text              
author="Stefan Schippers"
version=3.0
description="Verilog A model of a differential amplifier"
```

The first line `v {xschem version=.... file_version=... ` must always be present in .sch and .sym files as this information is used to recognize the file format.

georgtree, 2025-02-10:
ah, got it, that is really cool, thank you!

georgtree, 2025-02-14:
Hello!
is it possible to add ability to read symbol provided by symlink with `xschem getprop symbol symbolPath`?
`xschem place symbol` works with symlink, but not the getprop.
Thank you!

georgtree, 2025-02-14:
Example also doesn't work:
```
xschem [~/MDPSBLib] xschem getprop symbol comp_ngspice.sym
Symbol not found
while evaluating xschem getprop symbol comp_ngspice.sym
```

StefanSchippers, 2025-02-15:
xschem getprop works for loaded symbols

When a schematic is displayed all symbols used in the schematic are loaded in memory

To get attributes for on-disk symbols something must be added.

At the moment you can do this:

```
set current [xschem get schname] ;# save current schematic name
xschem load -noundoreset -nofullzoom -nodraw /path/of/xxxx.sym ;# load another symbol
xschem get schsymbolprop  ;# contains the type, format, template tokens
xschem get header_text ;# symbol metadata
xschem load -noundoreset -nofullzoom -nodraw $current
```

This will transparently load the specified symbol, do the queries, and reload the previous schematic.
It;s not a clen solution since you get out of the current schematic to load the symbol.
One advantage is that you can make any changes to the symbol and save it.

I have also added a new command:

xschem load_symbol /path/to/xxx.sym

xschem getprop symbol xxx.sym type
xschem getprop symbol xxx.sym format
xschem getprop symbol xxx.sym template

Symbols loaded this way do not interfere with the currently loaded schematic. However symbol metadata is not available in this case (it is not used in a schematic, so not loaded at all).

Another solution is to write a script that directly loads the symbol and gets the v {...} and K { ...} records, for metadata and global symbol attributes, respectively.

georgtree, 2025-02-15:
ah, got it, thank you, the only issue I see in that case is the necessity to properly preprocess the escaped sequences (like {} ), in case of xschem it is done automatically, but yeah I can do that. Thank you for the help and clarification.

closed 2025-02-15 by georgtree

georgtree, 2025-02-15:
Hello! Write script to extract tokens from header element, not sure if I took into account all corner cases (remove escaping only for \{ and \}). Maybe would be helpful for someone:
```tcl
proc symbolParse {file} {
    set file [open $file r]
    set data [read $file]
    close $file
    # pattern to find v {} and extract content inside brackets
    set pattern {(?i)v\s*\{((?:[^{}]|\\[{}])*)\}}
    if {![regexp $pattern $data -> content]} {
        return
    }
    # pattern to find name="value" match
    set pattern {(\w+)=(?:"([^"]*)"|(\S+))}
    set start 0
    while {[regexp -start $start $pattern $content -> name quotedValue unquotedValue]} {
        # Determine the value (quoted or unquoted)
        if {[string length $quotedValue] > 0} {
            set value $quotedValue
        } else {
            set value $unquotedValue
        }
        dict append tokens $name [string map {\\{ \{ \\} \}} $value]
        # Update the start index to continue searching after this match
        set start [expr {[string first $value $content $start] + [string length $value]}]
        # Safety check to prevent infinite loops
        if {$start <= 0} {
            break
        }
    }
    return $tokens
}
```

For symbol:
```
v {xschem version=3.4.5 file_version=1.2
description="Integrator element, the device is a part of XSPICE extension of Ngspice, more detailed
description could be find Ngspice manual"
pins="\{\{in in signal \{input of integrator\}\} \{out out signal \{output of integrator\}\}\}"
params="\{\{INOFFS 0.0 - \{Offset of input signal\}\} \{K 1.0 - \{Gain of input signal\}\}
\{LOWLIM -1e12 - \{Lower limit of output\}\} \{UPLIM 1e12 - \{Upper limit of output\}\}
\{LIMR 1e-6 - \{Upper & lower limit smoothing range\}\} \{OUTIC 0.0 - \{Output initial condition\}\}\}"
}
G {}
K {type=xspice_dig format="@name @pinlist @subckt OUTOFFS=@OUTOFFS K=@K OUTLOWLIM=@OUTLOWLIM OUTUPLIM=@OUTUPLIM LIMR=@LIMR" 
template="name=X1 subckt=differentiator OUTOFFS=0.0 K=1.0 OUTLOWLIM=-1e12 OUTUPLIM=1e12 LIMR=1e-9"}
S {}
E {}
L 4 20 0 30 0 {}
L 4 -20 -20 -20 20 {}
L 4 -20 -20 20 0 {}
L 4 -20 20 20 0 {}
L 4 -30 0 -20 0 {}
L 4 -30 0 -20 0 {}
B 5 -32.5 -2.5 -27.5 2.5 {name=in dir=in pinnumber=0 class=elec}
B 5 27.5 -2.5 32.5 2.5 {name=out dir=out pinnumber=1 class=elec}
T {du} -12.5 -9.375 0 0 0.15 0.15 { layer=7}
T {dt} -12.5 -0.625 0 0 0.15 0.15 { layer=7}
T {___} -12.578125 -8.828125 0 0 0.14 0.14 { layer=7}
T {K=@K } -20 20 0 0 0.1 0.1 {}
T {@subckt} -20 -40 0 0 0.15 0.15 {hide=true}
T {@name} -20 -30 0 0 0.15 0.15 {}
T {OUTOFFS=@OUTOFFS 
OUTLOWLIM=@OUTLOWLIM 
OUTUPLIM=@OUTUPLIM 
LIMR=@LIMR} -20 27.5 0 0 0.1 0.1 {hide=true}
```

it returns dictionary:
```
version 3.4.5 file_version 1.2 description {Integrator element, the device is a part of XSPICE extension of Ngspice, more detailed
description could be find Ngspice manual} pins {{{in in signal {input of integrator}} {out out signal {output of integrator}}}} params {{{INOFFS 0.0 - {Offset of input signal}} {K 1.0 - {Gain of input signal}}
{LOWLIM -1e12 - {Lower limit of output}} {UPLIM 1e12 - {Upper limit of output}}
{LIMR 1e-6 - {Upper & lower limit smoothing range}} {OUTIC 0.0 - {Output initial condition}}}}
```

StefanSchippers, 2025-02-15:
Thank you, i will try it. ( so I learn complex regexps!!) the characters `{`, `}` and `\` are escaped as `\{`, `\}`, `\\` when used as ordinary characters in side a {...} record. May be you can add \ if it is not already done.

My problem when parsing these fields was when a `\\ `(an ordinary `\` inside {...})  is placed before a closing brace `\\}`. In this case the closing brace marks the end of the {...} record.

StefanSchippers, 2025-02-15:
I have tested the symbolParse and it works very well. 

I have changed the first pattern to:
```
set pattern {(?:^|\n)v\s+\{((?:[^{}]|\\[{}])*)\}}
```
the changes are: 
- No case insensitive matching since the V {...} record exists and has a completely different meaning
- the `v` record tag must be at the beginning of a line (either beginning of file or immediately after a newline)
- there must be 1 or more spaces after the record tag and the opening brace `{`

Can I include this procedure into my xschem.tcl?

georgtree, 2025-02-15:
Hello, yes of course!

StefanSchippers, 2025-02-16:
I have integrated the procedure in xschem. Can you please validate it?

The following token=value assignments in the metadata (as written in the xschem GUI):
```
token0 = {stefan}
token1="stefan" 
token2="\\xxx\\ stefan \"schippers\" xxx"
```
That are saved in the file as:
```
v {xschem version=3.4.6 file_version=1.2
token0 = \{stefan\}
token1="stefan" 
token2="\\\\xxx\\\\ stefan \\"schippers\\" xxx"
}
```
Are correctly parsed. Note the optional spaces around the = and escaped escapes and escaped quotes inside quoted values.
The tokens are returned as:
```
version=3.4.6
file_version=1.2
token0={stefan}
token1=stefan
token2=\xxx\ stefan "schippers" xxx
```

georgtree, 2025-02-16:
Just tested, all works, thank you a lot!