StefanSchippers/xschem #310: Issue when connect to xschem from tcl script

georgtree, 2025-02-11:
Hello!
When I make netlist from gui, all works, but when I run xschem from tcl I got this:
```
** sch_path: /home/georgtree/MDPSBLib/examples/general_sim/linear_pendulum_ss.sch
**.subckt linear_pendulum_ss
*  X1 -  state_space_2nd  IS MISSING !!!!
*  X2 -  sgnl_const  IS MISSING !!!!
**** begin user architecture code

.tran 0.1 1000
.save v(y) v(x2.2) v(x2.1)
.control
    set filetype=binary
    run
    write linear_pendulum.raw
    set appendwrite
.endc

**** end user architecture code
**.ends
.end
```

Clearly there are missing components. I run xschem with `set channel [open "|xschem -r -x" r+]`.
Looks like it doesn't append the searching paths when run xschem from the script.

georgtree, 2025-02-11:
Sorry, I found the issue, the path was not normalized

georgtree, 2025-02-11:
Nope, looks like there is issue with loading xschemrc file...

StefanSchippers, 2025-02-11:
when xschem starts it load an xschemrc file in current directory if found or not found then in ~/.xschem/xschemrc.

When calling xschem from another script either:
- set the PWD of the child (xschem) process to the directory where the xschemrc is placed.
- call xschem with `--rcfile /path/to/xschemrc` 

In all cases ensure XSCHEM_LIBRARY_PATH is not built using $env(PWD) or [pwd] or other statements that depend on the place xschem i sstarted from.

georgtree, 2025-02-11:
In xschemrc: `append XSCHEM_LIBRARY_PATH :/home/georgtree/MDPSBLib/.sym_links/`. I load file as you stated above, still missing symbols.

georgtree, 2025-02-11:
Ah I am an idiot, sorry, problem solved, pointed to wrong location

closed 2025-02-11 by georgtree