StefanSchippers/xschem #311: little things:

olisnr, 2025-02-12:
1. if i insert a new part via the `instert-button` the new part is in the "move-mode", but the rotation hotkey `shift-r` dosnt work, while the contex-menu does.
if i inster a new part, and use the context-menu to set the allready moving part in the "move-mode" `shift-r` works.

2. often im in the wireing mode because i pressed `esc` only once. and then when i want to select something, i naturalmente am wireing this thing... would it be possible to change the color of the cross from yellow to red, if there is a drawing mode active?

3. the toggle command would be nice in the context menue

olisnr, 2025-02-12:
thanks a lot:
1. and 2. are working...

3. i would like to have the `Properties::Toggle...` or `shift-t` in the parts-context menu.

& in the newest version the plots with `autoload=0` still loads the graphs at startup. i think they shouldn't.

olisnr, 2025-02-12:
also the `Wave::Clear` does not clear the plots:

https://github.com/user-attachments/assets/21b3b71e-4c68-4ae0-9445-aa1ee0f27477

StefanSchippers, 2025-02-12:
The Clear Waves does the job, however graphs with autoLoad set will immediately load new data.

StefanSchippers, 2025-02-12:
To avoid automatic loading:

- do not set the autoload checkbutton.

or: 

- avoid setting a file name in the raw file: entry in graph dialog box. See video.

[image]

olisnr, 2025-02-13:
`autoload` is set to `0`, so the problem is the filename-override.

the thing is: in Xyce i cant have different simulations in the same `.raw` file. so i did some TCL to write different `.raw` for the different simulations, and for the different corners. as example:
```
VCO_schmitz_hampel_041_ac_mosff.spice.raw
VCO_schmitz_hampel_041_ac_mosfs.spice.raw
VCO_schmitz_hampel_041_ac_mossf.spice.raw
VCO_schmitz_hampel_041_ac_mosss.spice.raw
VCO_schmitz_hampel_041_ac_mostt.spice.raw
VCO_schmitz_hampel_041_mosff.spice.raw
VCO_schmitz_hampel_041_mosfs.spice.raw
VCO_schmitz_hampel_041_mossf.spice.raw
VCO_schmitz_hampel_041_mosss.spice.raw
VCO_schmitz_hampel_041_mostt.spice.raw
VCO_schmitz_hampel_041_tran_mosff.spice.raw
VCO_schmitz_hampel_041_tran_mosfs.spice.raw
VCO_schmitz_hampel_041_tran_mossf.spice.raw
VCO_schmitz_hampel_041_tran_mosss.spice.raw
```
i see no other option then specify the files...

StefanSchippers, 2025-02-13:
These are the possibilities. Let me know if you see a different usage mode.

https://github.com/user-attachments/assets/cc80fd8c-5a7b-4b87-9917-1b0101441941

olisnr, 2025-02-13:
i was thinking that `autoload=0` dont load the plot at startup. and in Your video it doesnt. but on my system the `autoload=0` is ignored see my videoclip again. i start a schematic from the file-browser, and it loads the graphs, even when they have `autoload=0`... and also the clear-command is ignored. 

`autoload=0` dont work for me for some reason i dont know. do You want to look at the files? or do You have an idea why Xschem ignors my autoload-flag?

[test_plot.zip](https://github.com/user-attachments/files/18791354/test_plot.zip)

StefanSchippers, 2025-02-13:
Most likely some graphs (the lower-left 4 graphs) have autoload=1 and once the raw file is loaded it is used by all other graphs.

olisnr, 2025-02-14:
ok, then if the files are loaded from an other plot, the autoload-flag has the meaning of reading a file into Xschem not loading a plot. its like there is an autoload-or for all plots using the same raw-file.

this wasnt clear for me. maybe it could be improved? plots are not loaded if autoload is zero? 

what do You think? i can live with it, but i think other could also misunderstand the meaning of autload.

StefanSchippers, 2025-02-14:
Yes I think I can change that. Will look at it tomorrow...

StefanSchippers, 2025-02-14:
> maybe it could be improved? plots are not loaded if autoload is zero?

There should be 3 options:
- graph is disabled (never show waves)
- graph is enabled and shows waves if the requested raw file has been loaded into memory
- graph is enabled and will trigger a forced loading in memory of the requested raw file, if not already loaded.

olisnr, 2025-02-14:
could the disabling and so on be done with the `shift-t`? it would feel natural.

closed 2025-03-01 by olisnr