StefanSchippers/xschem #312: Issue with substitution in model property

georgtree, 2025-02-12:
Hello!
I have element with next description:
```
name=A2 model=strans_id9926_@name device_model=".model strans_id9926_@name s_xfer gain=1.0 int_ic=[0.0] num_coeff=[1 0] den_coeff=[1 100]" 
```

After netlisting I got:
```
A2 pos net9 strans_id9926_A1
.model strans_id9926_A2 s_xfer gain=1.0 int_ic=[0.0] num_coeff=[1 0] den_coeff=[1 100]
```
Yesterday all worked fines, I think the issue in last bunch of commits.
Thank you.

StefanSchippers, 2025-02-12:
Hi George, I think it is fixed now. The regression due to a commit aimed to fix another testcase (and making yours fail again). Sorry for the trouble. Parameter resolution in the hierarchy (schematic? symbol? upper schematic? upper symbol? ...) is a minefield.

georgtree, 2025-02-12:
just tested, all works, thank you!

closed 2025-02-12 by georgtree