StefanSchippers/xschem #314: Windows building with MVSG and Visual Studio

georgtree, 2025-02-13:
Hello!
I am trying to compile xschem for windows, I did all steps, but the compilation failed, looks like because of the missing regex.h:

[image]

I did same building a few years ago, and didn't faced with that error.

Could you elaborate that please?

georgtree, 2025-02-13:
I tried both ready to use project and do it manually from empty one, issue is the same

StefanSchippers, 2025-02-13:
Hi, George, The windows maintainer is working on that. For the time being you can remove all the references to get_sp_cur. It's not an important thing. windows has no regexec/regcomp, will be replaced with a tcl call to regexp.

georgtree, 2025-02-13:
Yes, these errors dissapears but appeared news during linking:

[image]

StefanSchippers, 2025-02-13:
the `eval_expr.c` is a new file (allows evaluation of infix expressions). It is generated from a yacc file `eval_expr.y `and compiled with bison:

`bison -p kk -o eval_expr.c eval_expr.y`

This is another thing that must be included in the windows package build process. The windows maintainer was off for some time and is cacthing up.

You can use this file directly (the generated .c file).

You need to add the eval_expr.c file in the build process.

[eval_expr.zip](https://github.com/user-attachments/files/18790441/eval_expr.zip)

georgtree, 2025-02-13:
Next issue - missing header `#include <unistd.h>`

StefanSchippers, 2025-02-13:
I have removed this include line, it is not required there. Thank you.

georgtree, 2025-02-13:
Thank you, build was finished successful :)

closed 2025-02-13 by georgtree