StefanSchippers/xschem #317: Crash when open circuit

georgtree, 2025-02-17:
Hello!
I have circuit, and xschem crashes when I try to open it.

[all_symb.txt](https://github.com/user-attachments/files/18833092/all_symb.txt)

The output:
```
xschem [~/MDPSBLib] rename dir (null) to /tmp/xschem_emergencysave_all_symb_dfdddbdead failed
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave_all_symb_dfdddbdead

FATAL: signal 11
while editing: all_symb

```

I am sure there is a dump, but couldn't find it.
By removing lines I found that the crash is due to this line:
```
C {.sym_links/phys.mag.flux_tubes.shapes.fixed_shape/cuboid_rm_out.tcl("400 1.488 1200 3 12.5")} 200 -520 3 1 {name=X23 subckt=cuboid_rm_out NONLIN=1 MURC=1.0 L=0.01 A=0.01 B=0.01 MUI=400 CA=1200 CB=3 N=12.5 BMYMAX=1.488
hide_texts=true}
```

StefanSchippers, 2025-02-17:
Crashing bugs have the highest priority. 
If you can collect a testcase that reproduces the problem that would be great.
you can try to start xschem in debug mode: 

xschem -d 1 -l log.txt ......
and send the log.txt file...

georgtree, 2025-02-17:
here it is:
[log.txt](https://github.com/user-attachments/files/18833519/log.txt)

georgtree, 2025-02-17:
I provided wrong information, the line that causes crash is:
```
C {.sym_links/phys.mag.flux_tubes.shapes.hyst_magn/tellinen_soft_hyst_core_th.sym} 220 -200 0 0 {name=X1 subckt=tellinen_soft_hyst_core_th JS=1.8 BR=0.9 HC=120 K=1.0 L=0.01 A=0.01 MAGREL=0.0 EDDY=0 SIGMA=10e6 D=0.5e-3}
```

StefanSchippers, 2025-02-17:
Looks like it is loading a symbol (.sym file) containing  a component specifying a directory instead of a file:
l_s_d() is the function that Loads the Symbol Definition from file. It expects a file and gets this:

l_s_d() component: level=0, sym=**/home/georgtree/MDPSBLib/**, prop_ptr = (null)

Of course the program should not crash in these situations but reports the error or ignores the line. Will investigate. I suspect however the input is malformed.

georgtree, 2025-02-17:
You are right, I checked the file and there is a mentioned line:
```
v {xschem version=3.4.6 file_version=1.2}
G {}
K {type=magnetic_flux_tubes
format="@name @@n1 @@n2 @@t @subckt JS=@JS BR=@BR HC=@HC K=@K L=@L A=@A MAGREL=@MAGREL EDDY=@EDDY SIGMA=@SIGMA D=@D"
template="name=X1 subckt=tellinen_soft_hyst_core_th JS=1.8 BR=0.9 HC=120 K=1.0 L=0.01 A=0.01 MAGREL=0.0 EDDY=0 SIGMA=10e6 D=0.5e-3"}
V {}
S {}
E {}
L 16 -30 0 -20 0 {}
L 16 20 0 30 0 {}
L 16 2.5 -5 5 -5 {}
L 16 -5 5 -2.5 5 {}
B 5 -32.5 -2.5 -27.5 2.5 {name=n1 dir=inout class=flux_tubes}
B 5 27.5 -2.5 32.5 2.5 {name=n2 dir=inout class=flux_tubes}
B 5 27.5 17.5 32.5 22.5 {name=t dir=inout class=heat_tran}
B 16 -20 -10 20 10 {fill=0}
P 16 4 -5 5 -2.5 5 0 -5 2.5 -5 {bezier=1}
P 16 4 -2.5 5 0 5 2.5 -5 5 -5 {bezier=1}
T {L=@L 
A=@A
MAGREL=@MAGREL
EDDY=@EDDY} -20 10 0 0 0.1 0.1 {hide=false}
T {TS} 7.5 0 0 0 0.1 0.1 { layer=16}
T {@subckt} -20 -30 0 0 0.15 0.15 {hide=true}
T {@name} -20 -20 0 0 0.15 0.15 {}
T {JS=@JS 
BR=@BR 
HC=@HC 
K=@K 
SIGMA=@SIGMA 
D=@D} -20 37.5 0 0 0.1 0.1 {hide=true}
C {/home/georgtree/MDPSBLib/} 50 40 0 0 {}
```

I did not edited file in text editor, crash occurs when I edited symbol not sure if I can recreate that sequence of step.
The diff for file:

[image]

StefanSchippers, 2025-02-17:
I suspect this generator file name confuses xschem:

/home/georgtree/MDPSBLib/.sym_links/phys.mag.flux_tubes.shapes.hyst_magn/tellinen_everett_hyst_core.tcl

may be it confuses the first dot as the dot before the file extension (.tcl) and cuts the filename.

georgtree, 2025-02-17:
I posted the symbol, it is not a generator

StefanSchippers, 2025-02-17:
thank you. What is this line in the symbol supposed to do?
C {/home/georgtree/MDPSBLib/} 50 40 0 0 {}

It declares a component instance (strange, usually symbols do not have components, although they are allowed to).

But the weird thing is that the reference is a directory path

StefanSchippers, 2025-02-17:
Anyway it is a nice corner case to fix. Program should not crash because of bad/unexpected data.

georgtree, 2025-02-17:
I don't know, that is the issue :(
Edited properties of the text block, and then it crashed

StefanSchippers, 2025-02-18:
Ok more checks are now in place.
I have learned that it is possible to open a directory, the fopen() call does allow that, but you can't read anything from it (readdir() must be used). 
Now all file opening functions are checked for this.

If a symbol reference points to a directory it is handled correctly (willl show a missing symbol).

closed 2025-02-18 by georgtree