StefanSchippers/xschem #318: Rename vector (node) in loaded raw file

georgtree, 2025-02-18:
Hello!
I would like to ask about additional command that allows to change name of existing vector (node) in loaded raw file.
Something like:
```tcl
xschem raw rename oldNodeName newNodeName
```

I understand that there is a workaround - create new vector with desired name, and then copy the data from existing one and delete old, but it looks like too much overhead for renaming, especially if there are hundreds on large-sized vectors.
Thank you in advance!

georgtree, 2025-03-08:
Hello!
Is it possible to implement or it is better for me to find workaround for that? Thank you.

StefanSchippers, 2025-03-08:
Yes, that should be easy to do.

In the mean time you can do:
```
xschem raw add new_node {expr} 
xschem raw del old_node
```
`expr` can be just `old_node` or an expression, like 
```
old_node 2 *
```
If expr is a single word (like a raw node) braces are not necessary.

StefanSchippers, 2025-03-08:
The `xschem raw rename old_node new_node` is now implemented
If you have a chance please test.
Thank you

georgtree, 2025-03-11:
all works, thank you very much!

closed 2025-03-11 by georgtree