StefanSchippers/xschem #319: Graph object embedded into the symbol

georgtree, 2025-02-18:
Hello!
Is it possible somehow to attach graph rectangle to symbol?
The idea may sounds a little bit stupid, but sometimes I need small graph near the ammeter/node name, and it would be great to have such component which I can place and get the plot with corresponding vector already, instead of going to graph options and looking for specific vector to plot. Something like virtual oscilloscope in Multisim, but already a graph right in the schematic.

[image]

Thank you!

StefanSchippers, 2025-02-19:
I need to find a compromise solution. Embedding a graph inside a symbol means that the graph can be rotated / flipped as symbols do.

There are already 3 coordinate transformations when displaying a graph on screen.
- The graph X/Y numbers coming from simulation.
- the xschem 'world' coordinates (xschem uses a double precision infinite coordinate system). Each graph is mapped to a different area in xschem coordinates.
- the screen coordinates, that is the final pixels that show the graph.

I don't want to add a 4th transformation (graphs are interactive, can be zoomed / moved with the mouse. All needs to he handled differently if graph is rotated. That looks like a real nightmare and a source of numerous bugs.

Need to think of a simpler approach.

georgtree, 2025-02-19:
Hello, yeah I see, maybe is it possible to forbid rotation for graph objects? Or add special flag to forbid rotation of that symbol at all?
that make sense - there is no point in rotation of graph rectanlge

StefanSchippers, 2025-02-24:
@georgtree  I have a first implementation of this function. Currently there are 3 new symbols:

- scope.sym: Single ended voltage probe
- scope2.sym: Differential voltage probe
- scope_ammeter.sym: current measure probe (ammeter)
See video.

https://github.com/user-attachments/assets/1c80fd9c-d93b-41c0-97c0-3f12171e0208

georgtree, 2025-02-24:
Yeah, that is what I meant!
Now the level of possible visualizations/annotations is at the unbeatable level (I don't know  commercial software with such
graphs right IN schematic, even Keysight ADS doesn't have that).
Thank you a lot!

georgtree, 2025-04-30:
Hello!
When I hover over the graph object it does not change the control mode, as it does with usual graphs.

StefanSchippers, 2025-04-30:
This should be fixed. In the mean time if you hit ctrl-key down you will enter graph edit mode.

StefanSchippers, 2025-04-30:
I have checked. It is deliberate to require Control key down to operate on graphs inside the scope.sym symbol.
This is done to allow click and drag of the symbol as a whole (the symbol and the embedded graph move together).

To operate on the inner graph press the Control key.  button click + drag will move the graph waveforms and not the symbol.

https://github.com/user-attachments/assets/300f78e7-80a4-4d4b-b2d2-8d8a9e795b3b

closed 2025-05-05 by StefanSchippers