StefanSchippers/xschem #321: Menus blink and flicker

itsayellow, 2025-02-19:
I'm using xschem (commit e9c927d8, but I believe this was still happening on the latest release) inside a Docker container with Centos 9 stream, and then sending to XWindows on a Mac using XQuartz as the X server.

I've always been seeing this problem where the menus and menu items blink on and off (text and empty space where text was) as I move the mouse over menus. Sometimes, after I've been using the program for a while, the menubar items start blinking really rapidly as if they are reprinting over and over again.

One thought: Is it possible I need to compile differently, or using different X libraries?

Here is a screen recording of the "slow blink" where mousing over items in a menu turns some on or off.
https://drive.google.com/file/d/1BfdFyf4kaoPrG7SvtCNF6rUJP_9xiZfi/view?usp=sharing

I couldn't capture the fast blink behavior, because it only happens sporadically.

StefanSchippers, 2025-02-19:
Looks like it is a rendering problem on the Xserver. I have 4 different laptops, from very old and small with atom processors up to corei7. Non none of these machines I see this issue, even in cross configurations (user on one machine, xschem running on another machine).

Try to go into the XQuartz preferences and try to disable any hardware acceleration if possible.

itsayellow, 2025-02-19:
I looked but I can't find any acceleration options for XQuartz.

It's weird because I don't see glitches like this with other X programs.

Is there possibly some compile switch I could alter that might change something with the menu display?  Or might I be missing some optional dev library that causes the compilation process to fall-back to some lesser library?

StefanSchippers, 2025-02-19:
Ok try to start this application: 
`/usr/share/doc/tk8.6-doc/demos/widget`
If you don't have it install the tk8.6-doc package. (it is useful anyway as it contains the man pages).
start it with:
`wish /usr/share/doc/tk8.6-doc/demos/widget`
or 
`wish8.6 /usr/share/doc/tk8.6-doc/demos/widget`
(or whatever tcl-tk version you have on the system).

see details in video.

https://github.com/user-attachments/assets/5528a251-cbe8-4c20-b9bc-4f90c2ceec08

itsayellow, 2025-02-19:
Centos 9 inside of Docker with XQuartz on mac: OK, I guess it's a tk thing, possibly with the combination centos, or xquartz.  I see the same issue with the menus in the tk menu demo.  Certain pieces of text in the menus disappear and reappear on mouse-over.  Sometimes menu label, sometimes hotkey description, etc.

Ubuntu 24.04: I tried the same demo on Ubunt 24.04 inside of a Parallels virtual machine, and it seems to work ok.  I don't see the menu items occasionally go blank.  I do see the whole menu bar blink as I mouse to a different menu, so maybe there's some particular mouseover behavior that is a little funny that goes horribly wrong in certain implemenatations.

For other readers, on my Centos 9 system, you can find widget at path `/usr/share/tk8.6/demos/widget` installed as part of the regular 'tk' package.

StefanSchippers, 2025-02-19:
Looks like some Expose events get lost and the toolkit (Tk) does not redraw the image.

I see other issues with the same problem:
https://github.com/XQuartz/XQuartz/issues/423

itsayellow, 2025-02-19:
Sadly, the last commit to XQuartz is over 2 years ago, so it's not really being maintained.  It may be time to find a new Xserver for macOS

StefanSchippers, 2025-02-19:
May be you could try the 'macports' way as described here:
https://www.xquartz.org/releases/index.html

hpretl, 2025-05-05:
@itsayellow @StefanSchippers Yes, XQuartz on macOS has issues. I also have the menu issue, in Xschem as well as Magic. By now I got used to it, apparently, but if you find a fix that would be great.

itsayellow, 2026-08-18:
I'm using the newest betas of XQuartz (XQuartz is being updated again!) and these issues seem to be completely fixed.  I also think the stable version released this year (XQuartz 2.8.6) also fixes things.

I think this is "solved" on macOS with the latest stable XQuartz.

hpretl, 2026-10-29:
I agree, I have no issues with Xschem lately, seems XQuartz fixed the issue. 

We could close this.

StefanSchippers, 2026-10-29:
Very happy to close this issue, since there is not much I can do (as far as I know) about it.