StefanSchippers/xschem #323: Netlisting changes wire names

kwmartin, 2025-02-20:
[Sandbox](https://drive.google.com/uc?export=download&id=1q_yxz8rExzIbrluf_O6qV42aiC0yNWcH)
sandbox unzips into a single directory called sandbox. >cd sandbox; xschem
The sandbox is self contained (as tested).
The schematic to open is std_lib/km_tb_lib/sch/TST_DIG.sch
```
v {xschem version=3.4.6 file_version=1.2
* xschem is copyright 2022 Stefan Frederik Shippers
* converted from Viewdraw V51
}
G {}
K {}
V {$TB_BODY}
S {}
E {}
N 50 60 60 60 {lab=VDD,VSS}
N 40 40 60 40 {lab=CLK}
N 120 40 180 40 {lab=A1A}
N 170 60 180 60 {lab=VDD,VSS}
N 240 40 320 40 {lab=A1B}
N 380 40 400 40 {lab=A1C}
N 310 60 320 60 {lab=VDD,VSS}
C {devices/lab_wire3.sym} 50 60 0 0 {name=g3 lab=VDD,VSS pinnumber=1}
C {devices/lab_wire2.sym} 40 40 0 0 {name=g5 lab=CLK pinnumber=1}
C {devices/lab_wire3.sym} 170 60 0 0 {name=g11 lab=VDD,VSS pinnumber=1}
C {devices/lab_wire2.sym} 400 40 0 0 {name=g16 lab=A1C pinnumber=1}
C {devices/lab_wire3.sym} 310 60 0 0 {name=g17 lab=VDD,VSS pinnumber=1}
C {diglib28/sym/INV1_1X1.sym} 60 10 0 0 {name=INV1 MODEL=DG_INV1_1X1 spiceprefix=X}
C {diglib28/sym/INV1_1X1.sym} 180 10 0 0 {name=INV58 MODEL=DG_INV1_1X1 spiceprefix=X}
C {diglib28/sym/INV4_1X1.sym} 320 10 0 0 {name=INV64 MODEL=DG_INV4_1X1 spiceprefix=X}
```
Note: ```N 120 40 180 40 {lab=A1A}``` is labeled A1A
Open schematic, select wire after first inverter, >q, and note it is named A1A
view netlist (<shift>A), change to spice netlist option, netlist circuit, note in netlist (stored in sandbox/netlists:
```
**.subckt TST_DIG
XINV1 CLK net1 VDD VSS INV1_1X1
XINV58 net1 net2 VDD VSS INV1_1X1
XINV64 net2 A1C VDD VSS INV4_1X1
**.ends
```
The wire A1A is now called net1, and A1B is now called net2. If the wire in the schematic is examined, it is now called: #net1.

[image]

If the schematic is first saved, and then std_lib/km_tb_lib/sch/TST_DIG.sch is examined,
the relevant lines are now:
```
N 120 40 180 40 {lab=#net1}
N 170 60 180 60 {lab=VDD,VSS}
N 240 40 320 40 {lab=#net2}
N 380 40 400 40 {lab=A1C}
N 310 60 320 60 {lab=VDD,VSS}
```
A link to the new schematic file is: [TST_DIG.sch](https://drive.google.com/uc?export=download&id=1q_yxz8rExzIbrluf_O6qV42aiC0yNWcH). I'm guessing the problem is somewhere in netlist.c. maybe around: wirecheck() or name_attached_nets(). If you could tell me how to get gdb attached to trace netlisting, I can try and track it down (do I need to add a loop somewhere to allow for attaching and after attaching breaking out for tracing - where should I place such a loop?).

Somewhat related (more likely a new issue). In schematics, the symbol properties (example: "name=INV1 MODEL=DG_INV1_1X1 spiceprefix=X") are included in lines such as:
```
C {diglib28/sym/INV1_1X1.sym} 60 10 0 0 {name=INV1 MODEL=DG_INV1_1X1 spiceprefix=X}
```
inside curly braces. It's similar for specifying wire labels. In both cases, it would be nice to be able to control visibility, font size, attachment point, etc. A possible approach would be to make the {} parameter specification hierarchical. For example, when the defaults are not desired, an additional sub-parameter can be included, such as:
```
C {diglib28/sym/INV1_1X1.sym} 60 10 0 0 {name=INV1{visible=true size=5 attach=3} MODEL=DG_INV1_1X1 spiceprefix=X}
```
This shouldn't break any existing data bases, as it would only be added when desired. Just a thought. Let me know if you think it is worth a discussion (on discord? or sorceforge?).
Overall, a very nice schematic capture program, Thanks.

StefanSchippers, 2025-02-20:
Nets without names are auto-named by xschem when netlisting. If you want a consistent naming generate a netlist **_before_** saving the file.
Nets without a label are not guaranteed to retain their names, expecially if the circuit is changed.

StefanSchippers, 2025-02-20:
> inside curly braces. It's similar for specifying wire labels. In both cases, it would be nice to be able to control visibility, font size, attachment point,

There are some functions to control the symbol appearance, ash shown here.

https://github.com/user-attachments/assets/04de37a4-01d7-4807-bec4-5781126ca468

kwmartin, 2025-02-20:
Is it possible to hide only some of the attributes, not all of them?

kwmartin, 2025-02-20:
Nets without names are auto-named by xschem when netlisting. If you want a consistent naming generate a netlist before saving the file.

Not sure what this means. It seems you have to have a saved file before it is possible to generate a netlist? So one needs a file; then needs to netlist, and then needs to save the file a second time?

Nets without a label are not guaranteed to retain their names, especially if the circuit is changed.

Also do not set names directly in wires. (like doing a 'q' on a wire and changing the 'lab' attribute).

If I understand correctly, doing a q on a wire and changing the 'lab' attribute is not considered to be naming a net; to name a net, one must add a lab_pin.sym or a lab_wire.sym. Thank you, I did not know this. Does doing a q on a wire and changing the 'lab' attribute have any effect?

StefanSchippers, 2025-02-20:
> Is it possible to hide only some of the attributes, not all of them?
Some symbol attributes may have the `hide=instance` flag. These texts are not displayed when the symbol is instantiated, however are visible if you descend into the symbol.
All texts will be displayed if you enable `View->Show/Hide->Show hidden texts`

[image]

StefanSchippers, 2025-03-14:
> Not sure what this means. It seems you have to have a saved file before it is possible to generate a netlist? So one needs a file; then needs to netlist, and then needs to save the file a second time?

xschem does not need to save a schematic to generate a netlist.
If you draw a new schematic with lot of wires and no labels on them, and you want to know what the name of a wire will be (the one you get by doing 'q' on a wire and reading the 'lab' attribute) press the 'Netlist' button. This ensures the circuit is checked and all wires are consistently named.

closed 2025-03-19 by StefanSchippers