StefanSchippers/xschem #324: Xschem on windows using the Cygwin  - Cygwin/X Posix layer

StefanSchippers, 2025-02-20:
It is possible to run xschem (as well as many unix console / graphical applications) on WIndows by installing the cygwin library.

Cygwin is a DLL that provides a translation layer for almost all Unix system calls and many Unx tools (like flex, bison, awk, seda bash shell and many more).
Cygwin/X provides an X server built on top of cygwin that provides all the X11 functionality.

fschrive, 2025-02-21:
Here is a step-by-step guide to installing Cygwin and Xschem:

**1) Go to https://cygwin.com/ and install Cygwin (click on setup-x86_64.exe)**
- Open setup-x86_64.exe > Next > Install from Internet > Next > Root Directory = C:\cygwin64, Install For = All Users (RECOMMENDED) > Next > Local Package Directory = C:\Users\your_name\Downloads > Next > Use System Proxy Settings > Next > Choose a Download site > Next
- Select Packages > View = Full > Search = wget > Select Package = wget (latest version from the drop-down menu) > Next > Next (wait for the Cygwin setup to complete)

**2) Run Cygwin (double-click on the Cygwin icon on Desktop) and install the necessary packages:**
- Install apt-cyg (type or copy/paste the following commands)
_wget rawgit.com/transcode-open/apt-cyg/master/apt-cyg
install apt-cyg /bin_
- Install git, make, cmake, extra-cmake-modules, gcc-core and gcc-g++ (type or copy/paste the following commands)
_apt-cyg install git make cmake extra-cmake-modules gcc-core gcc-g++
apt-cyg install libcurl4 libbrotlidec1 libbrotlicommon1 libgsasl18 libidn12 libntlm0 libnghttp2_14 libssh2_1 libguile3.0_1 libffi8 libgc1 libmpc3 libisl23 cygwin-devel binutils w32api-runtime m4_
- Download the Xschem sources from github.com (type or copy/paste the following commands)
_git clone https://github.com/StefanSchippers/xschem.git xschem-src_
- Install libX11_6, libXrender1, libxcb1, tcl, tcl-tk, flex and bison (type or copy/paste the following commands)
_apt-cyg install libX11_6 libXrender1 libxcb1 tcl tcl-tk flex bison
apt-cyg install libXau6 libXdmcp6 libXft2 libXss1 libfontconfig1 libexpat1 libfreetype6 libXext6 libpng16 libX11-devel libX11-xcb1 libX11-xcb-devel libXrender-devel libXpm4 libXpm-devel xorgproto tcl-devel tcl-tk-devel_
- Check that all the required libraries are installed (type or copy/paste the following commands). If everything goes well, the message “Configuration complete, ready to compile” should appear
_cd xschem-src
./configure_
- Compile Xschem (type or copy/paste the following commands)
_make_
- Install Xschem (type or copy/paste the following commands)
_make install_

**3) Install any X server for Windows (example below for VcXsrv)**
- Download and install VcXsrv at https://sourceforge.net/projects/vcxsrv/
- Run VcXsrv (double-click on the XLaunch icon on Desktop) > Multiple windows > Next > Start no client > Next > Clipboard = yes, Native opengl = yes > Next > Finish

**4) Go back to Cygwin and run Xschem (type or copy/paste the following commands)**
_cd /usr/local/bin/
export DISPLAY=:0
./xschem_

Hope this helps.
Fred.

georgtree, 2025-02-28:
I can confirm that it works, and the more surprisingly, the performance is better than under WSL2 with Xwayland.