StefanSchippers/xschem #325: Feature Request: Opening multiple .sch files at once

simi1505, 2025-03-06:
Hi @StefanSchippers!

Is it possible to open multiple .sch files at once in the "Load file" window?

For example, if I want to open all testbenches of my amplifier core (see figure), I would normally press CTRL and select the files with a left mouse click. Unfortunately, this is not working in Xschem. Is it possible to implement this in Xschem?

[image]

Thanks and best regards

Simon

StefanSchippers, 2025-03-06:
You can invoke xschem from the command file with multiple files. It will open each file in a different window / tab.
`xschem testbench1.sch testbench2.sch testbench3.sch`
The file load widget does not support multiple selections (with ctrl-click). May be I will add that.
You can also add a `launcher.sym` in the first testbench and set the following command:
```
tclcommand="
  xschem load_new_window testbench2.sch
  xschem load_new_window testbench3.sch
  xschem load_new_window testbench4.sch
"
```
You ctrl-click on it and it will load the specified testbenches in different windows or tabs.

StefanSchippers, 2025-03-09:
Latest commit allows multiple selections when loading files.
Please test, needs some validadion!