StefanSchippers/xschem #326: questions

olisnr, 2025-03-06:
for the newest version of Xschem i would like to know:
1. how to draw diagonal wires.
2. what does the  `set snap_cursor` in the xschemrc. i saw no difference.
3. what is the meaning of `set big_grid_points`? i was expecting to be able to make the grid-points bigger like 2x2 or 3x3 pixels. i would be nice on my cheep 2nd-hand 4k display that dont show single pixels that bright like lines.

StefanSchippers, 2025-03-06:
1. press space bar 2 times after starting a wire.

2. snap_cursoer si part of a set of options to emulate cadence:Try:
set cadence_compat 1
set infix_interface0
set snap_cursor 1
Then try the 'w' keybinding. Use 's' to start a wire ftom the shown diamond shaped cursor. Use 's' again to end a wire to the shown diamond cursor. Use space bar to toggle wire orientations.

3. set big_grid_points 1 will make the grid points bigger as you zoom in.

To set bigger grid and lines for 4K+ displays you can try:
set line_width 2
set change_lw 0
set big_grid_points 1

The second line disallows changing line width when zooming in / out.

olisnr, 2025-03-07:
thanks a lot.

`cadence_compat 1` changes the behavior much. im not able anymore to select first and move then. i have to make the move command and select then. is this correct?

`set line_width 2` does change also all lines (like the name implies :) but i only want to change the to little dots. the lines are visible perfect with 1 pixel wide. if its possible to set the grid-points only, it would be interesting. You draw circles. it could also be squares or little crosses. it only should be bigger then 1x1pixel...

when trying the `Options::Crosshair::Crosshair size` using ≠ 0, i get a square as cursor. is this intended?

[image]

[image]

StefanSchippers, 2025-03-07:
> set line_width 2 does change also all lines (like the name implies :) but i only want to change the to little dots. the lines are visible perfect with 1 pixel wide. if its possible to set the [...]

This needs to be added. Separate widths for dots.

> 
> when trying the Options::Crosshair::Crosshair size using ≠ 0, i get a square as cursor. is this intended?

yes, some users want to see a little square snapped to the nearest grid point. If you press Shift the square will move to the closest instance pin or wire endpoint. Pressing Shift-W (instead of w) will start a wire from that point.

In Cadence mode you click to select an object, press 'm' for move then click again to start moving, click again to set the final position. Some users like this extra click. I don't.

olisnr, 2025-03-07:
thanks. 

there is one more thing i would like: a simple notification if there is a **simulation sub-process** running. 

at the moment i open the `Simulation::List running sub-processes` window and stick it to top. 
this is ok, if i have only one open Xschem, otherwise i dont know what schematic is the parent of the process-list.
the 2nd problem with this solution is: You need really to look what the processes are, it could be also an editor...

can we have something like the Simulate-button shows the number of current running simulation processes if >0? or a message in the info-bar at the bottom?

StefanSchippers, 2025-03-07:
There is now a new variable: grid_point_size.
if you add in xschemrc:
set grid_point_size 2
a 2x2 pixel square is drawn for each point.
you can set it also in the menu View->Set grid point size

If you use an even number of pixels (2x2) it can not be centered of course, but on a UHD display you will barely see that.

[image]

olisnr, 2025-03-07:
great! i use 3...

about the process-list question. in principle the coloring of the `Simulate-button` does something like this. but it is updated at ending of every simulation process. so it dont show if a simulation is still running. and this is what i need to know.

StefanSchippers, 2025-03-15:
> about the process-list question. in principle the coloring of the `Simulate-button` does something like this. but it is updated at ending of every simulation process. so it dont show if a simulation is still running. and this is what i need to know.

I think you should store the ID of the simulation in a variable:

`set a [simulate]`

when simulation is finished you can query $execute(exitcode, $a)
if this variable does not exists then simulation is running

 `catch {set execute(exitcode,$a)} res`

This command returns 1 if simulation is running and res will be: 
`can't read "execute(exitcode,1)": no such element in array`

when simulation finishes the command returns 0 (and res will contain the exitcode of the simulation, usually 0)

olisnr, 2025-03-17:
thanks, the simulation command in the launcher does this. but i try to show the number of running simulations outside of it. i think i will use the `proc get_running_cmds {} {` as source.

while trying to learn from to xschem.tcl i saw the function `proc from_eng {i} {` that seems to convert spice-value metric prefixes to numbers. in Xyce there are more then in the function. may be You want to include the `x` as alternative to `meg`:

[image]

olisnr, 2025-03-17:
would it make sense to change `tctx::[xschem get current_win_path]_simulate_id` from a value to a list? then it would be easy to manage many concurrent running simulations. 
this could be interesting because of ngspice simulations cant have different corner-libs in the same simulation, and Xyce can have only one simulation-type per netlist file. so starting different simulations could also be interesting for others.
what do You think?

olisnr, 2025-03-18:
now i try to customize the `processlist-dialog` in a `launcher.sym`.
1. it should show the schematics its connected to:
```
proc list_running_cmds {} {
  ...
  wm title .processlist [file rootname [file tail [xschem get current_name]]]
  insert_running_xyces $lb
}
```

2. it should show the simulations only:
and here im not shure what way to go. i can write a `proc insert_running_xyces {lb} {` as replacement for the 
`proc insert_running_cmds {lb} {` but then i miss the updates from events. 

what would be the best way to filter the processlist?

olisnr, 2025-03-18:
i wasnt aware, that if i over-write `insert_running_cmds` it will also called from the event-procedures...

so my problem is soved :) but may be You will change the title of the process-list window in the future. and may be add a regex filter text-field to the dialog?

StefanSchippers, 2025-03-18:
@olisnr I was a litle bit busy, so I am now trying to catch up with your latest news here ....

StefanSchippers, 2025-03-18:
> You will change the title of the process-list window in the future. 
This can be done, however not with the title of the schematic window. The process list is global. If there are multiple schematic windows open, each with its set of simulation (perhaps multiple in parallel on process corners) the list shows all these processes.

olisnr, 2025-03-18:
my way of using Xschem is at the moment, there are typical more then one Xschem processes running. every one with an other schematic or version. the simulations are started by a `launcher.sym ` and i open a processlist for every Xschem process that has something to simulate.
the simulation commands are very long, and most of the time there is nothing in the list anyway. so its not practical to find the corresponding Xschem to a processlist from the entries of the list.

thus i would prefer to see the TOP-level schematic name in the titelbar.

StefanSchippers, 2025-03-18:
For multiple simulatons I think we already discussed this approach: Loading a schematic, launching a first simulation, changing some parameters, changing output file name, running another sim and so on.
in this example I am collecting all the **id**s into my **id_list**

```
xschem load poweramp.sch
set id_list {}

xschem netlist
set id [simulate]
# some delay to allow ngspice to start and read the netlist before changing it
delay 1000
lappend id_list $id

# change VPP, output file name and input signal amplitude
set commands [xschem getprop instance NGSPICE value]
regsub {\.param VPP=50} $commands {.param VPP=30} commands
regsub -all {poweramp\.raw} $commands {poweramp1.raw} commands
xschem setprop instance NGSPICE value $commands
xschem setprop instance V3 value {dc 0 sin 0 0.5 {frequ} 1m}

xschem netlist
set id [simulate]
# some delay to allow ngspice to start and read the netlist before changing it
delay 1000
lappend id_list $id

# change again VPP and output file name
regsub {\.param VPP=30} $commands {.param VPP=25} commands
regsub -all {poweramp1\.raw} $commands {poweramp2.raw} commands
xschem setprop instance NGSPICE value $commands

xschem netlist
set id [simulate]
# some delay to allow ngspice to start and read the netlist before changing it
delay 1000
lappend id_list $id

# poll loop
while {1} {
  delay 1000
  set busy 0
  foreach id $id_list {
    set var "execute(cmd,$id)"
    if {[info exists $var]} { set busy 1}
  }
  if {!$busy} { break }
}
puts Done
```

StefanSchippers, 2025-03-18:
> thus i would prefer to see the TOP-level schematic name in the titelbar.

Ah, Ok, this makes sense.

olisnr, 2025-03-18:
yes the `launcher.sym` code You gave made work long time ago, is still working perfect. 

my current topic is only monitoring the simulations, seeing, what simulation has problems, or needs long time. typical there is a Xyce started for .op .dc .ac and .tran at the same time.

StefanSchippers, 2025-03-18:
ok the title now is: 

[image]

I have specified 'started from' since it is not relevant only for that schematic. the schematic may change after opening the process list or you might have multiple windows/tabs for the same xschem process.

olisnr, 2025-03-18:
thanks. 
one question. is the title updated if the schematics is "saved as"? 
i would like to have the title updated, even if "started from" isnt correct then. i would make it like 
`child processes: $top_schematics_name`
but maybe thats problematic for tabbed Xschem or cases, where simulations arent in the top-schematic

StefanSchippers, 2025-03-18:
ok I have modified the title. It shows the schematic open in the active window or tab and is updated if you Save-As or open a different schematic or switch to another tab. Was not difficult. Updates are done at 1 sec frequency to avoid too much background activity.

https://github.com/user-attachments/assets/34747fd9-d210-4ba7-9815-e0b37a9c8011

olisnr, 2025-03-18:
thanks a lot!