StefanSchippers/xschem #327: crash after 2nd time trying 'Waves::Op Annotate'

olisnr, 2025-03-07:
in the newest version of Xschem i can provoke an crash with trying to load .raw-data two times. Xschem says:

```
ich@P43s:~$ xschem
Using run time directory XSCHEM_SHAREDIR = /usr/local/share/xschem
Sourcing /usr/local/share/xschem/xschemrc init file
Sourcing /home/ich/.xschem/xschemrc init file
SG13G2_MODELS: /home/ich/share/pdk/ihp-sg13g2/libs.tech/ngspice/models
SG13G2_MODELS_XYCE: /home/ich/share/pdk/ihp-sg13g2/libs.tech/xyce/models
% error: get_sim_code{}; simulation not implemented op

free_rawfile(): clearing data
error: get_sim_code{}; simulation not implemented op

Raw file data read: /ALL/Xschem/VCO.xyce/simulation/VCO_HBT_RO90_TB_001_vco_mostt.spice.raw
points=9, vars=283, datasets=1 sim_type=dc
Raw file data read: /ALL/Xschem/VCO.xyce/simulation/VCO_HBT_RO90_TB_001_vdd_mostt.spice.raw
points=9, vars=283, datasets=1 sim_type=dc
error: get_sim_code{}; simulation not implemented op

free_rawfile(): clearing data
raw_read(): no useful data found
extra_rawfile() read: /ALL/Xschem/VCO.xyce/simulation/VCO_cheep_HBT_TB_001_mostt.spice.raw not found or no "op" analysis
Raw file data read: /ALL/Xschem/VCO.xyce/simulation/VCO_cheep_HBT_TB_001_mostt.spice.raw
points=1, vars=159, datasets=1 sim_type=dc
error: get_sim_code{}; simulation not implemented op

rename dir (null) to /tmp/xschem_emergencysave_VCO_HBT_RO90_TB_001_fdbdbcgfaf failed
EMERGENCY SAVE DIR: /tmp/xschem_emergencysave_VCO_HBT_RO90_TB_001_fdbdbcgfaf

FATAL: signal 11
while editing: VCO_HBT_RO90_TB_001
```

the files are: 

[VCO.xyce.zip](https://github.com/user-attachments/files/19126156/VCO.xyce.zip)

olisnr, 2025-03-07:
it also can crash with the correct .raw file

olisnr, 2025-03-07:
so its easier to reproduce: [VCO.xyce.zip](https://github.com/user-attachments/files/19126523/VCO.xyce.zip)

https://github.com/user-attachments/assets/5a753a70-5082-4bef-a6f8-07420834b794

StefanSchippers, 2025-03-07:
Can you try to annotate / load raw files manually, without any embedded launcher / script?
does it crash?

olisnr, 2025-03-07:
yes, but it crashes only the 2nd time. in the 1st post there is the command line output of Xschem in this case.

https://github.com/user-attachments/assets/a4ebe730-5a0a-4b79-aaab-02559ba0aa89

StefanSchippers, 2025-03-07:
Ok I get the crash, ready to fix it (hopefully).

olisnr, 2025-03-07:
did You see that Xyce seem to have .raw file problems?
https://github.com/Xyce/Xyce/issues/132

olisnr, 2025-03-07:
super! thanks!!!

closed 2025-03-07 by olisnr