StefanSchippers/xschem #328: strange lines in plots at absurd zoom-levels

olisnr, 2025-03-07:
if i zoom in a plot very deep i see wrong lines:

https://github.com/user-attachments/assets/2f0720d0-7367-4aa7-9936-874b966ac359

StefanSchippers, 2025-03-07:
Yes this is known. If you zoom in too much the integer coordinates of the lines go beyond the 16 bit limits of the Xlib library and wrap around the universe. Some clipping could be added but this slows down the rendering since graphs typically contain thousands of points.

olisnr, 2025-03-07:
ok... then its intended :)

olisnr, 2025-03-07:
can You tell me what 16bit limit is the problem?

StefanSchippers, 2025-03-07:
xschem uses double precision numbers for schematic coordinates. This means the available space for a schematic is extremely big while keeping high resolution (16 significant digits).

When zoomed into a small portion of the schematic the xschem coordinates (double precision 64 bit numbers) must be translated into xlib pizel coordinates, described with 16 bit signed integers. So the range for X/Y screen coordinates is  -32,768 to +32,767. ( in practice only 0 to +32767 is used),

If the screen size is 1920 x 1080  only X/Y coordinates that are below these values will be visible.
Everything that maps to coordinates outside the screen will not be visible and is "clipped".

However consider this scenario

xschem has a wire that goes from 0,0 to 100000,100000
the 0,0 xschem coordinate is centered in the top -left corner of the screen and zoom level is 1 (one xschem unit corresponds to 1 pixel)

Only the portion of 0,0 - 1080,1080 of the line will be visible on screen, the rest is clipped.

If you convert the floating point coordinates to 16 bit integer, above 32767 the integer will overflow and jump from max positive (32767) to lowest negative (-32768). SO you see these segments wrapping around and running through the screen multiple times.

For circuit elements (wires, lines, symbol objects) xschem does a software clipping before converting floating point to integers, so screen coordinates will never overflow and you won't see garbage.

For graphs only xlib clipping is done ( the graph rectangle is set as a clip region, anything outside is clipped autmaticlly by the xserver, however no sw clipping when converting double to int is done. So as you saw at absurd zoom-in levels you will see coordinate aliasing.

StefanSchippers, 2025-03-08:
@olisnr  I think last commit addresses this problem. Can you please test (You have a 4k display so you see this problem better). I found a cheap way to clip values that are too far off screen, so it should not affect performance.

olisnr, 2025-03-09:
You fixed the strange lines, but its still somehow strange: at nihilistic absurd zoom levels the time seems to change. see the CPU load-plot. there is no CPU load, but the plot stucks. and then, after some time there comes grey and green color out of the black void. 
the plot stuck a s9 and updates at s12 and s18. i was zomming in via mouse-wheel the hole time. at this s18 there is the color of the grid points, but it shoudent. next plot-update is at s26. the color is the void again. then the color changes to point and then to plot-line. and the plot is not every time updated. but see Your self:

https://github.com/user-attachments/assets/407ff1db-1037-48eb-ae31-535af3b0a012

olisnr, 2025-03-09:
i think its ok. but something still gets wild.

by the way: i think there is no reason for the cursor to have a grid in the plot. i would prefer no grid and instead of the coordinates to see the XY-values inside the plot in the info-field at the bottom. what do You think?

StefanSchippers, 2025-03-09:
I see you are zooming in with variable line width. 
At super zoom-in levels the width of lines becomes enormous.
I think I have not set a software limit on this (and may be I will do in a next commit)
Drawing super wide lines is slower and slower and may be the reason for the problems you see
Try to set a fixed line width (View-> Set line width)  and see if the same problem occurs.

olisnr, 2025-03-09:
yes, a zoom limit would solve strange behavior. 

for the cursor grid in the plot i have a video. if i want to measure as example the frequency via cursor, the snapping isnt helpfull. snapping is anyway only interesting to draw something. also the coordinates in the schematics arent really interesting if You are in a plot:

https://github.com/user-attachments/assets/0ca84fd2-4a18-4158-90f8-606b486dcf74

olisnr, 2025-03-09:
and here comes the video using line with 1. it has only one artifact:

https://github.com/user-attachments/assets/ae271f99-c00a-4902-9e21-f488012bdc9f

StefanSchippers, 2025-03-09:
You can view x/y values by enabling the meter (Ctrl-m inside the graph).
When pressing control  key the crosshair is cleared, movement is not snapped ans the meter shows the graph coordinates.

https://github.com/user-attachments/assets/b213d518-d9b4-4b08-899a-d742167dd9b0

olisnr, 2025-03-09:
super thats what i need. i only think its a bit to complex to use. why is there a need to press `esc` to disable the snapping and why there is the need to change the cursor? wouldnt it be more natural to remove the snapping over a plot, and also show the XY values if the flag, manipulated by `cntr-m` is enabled?

in the moment, if i press 'esc' in the `show-XY-mode`, i will see the XY window only after moving the little cursor, or after pressing `cnrt-f`. simply pressing `esc` does only change the cursor to little. may be thats because of a cinnemon-setting :)

StefanSchippers, 2025-03-09:
If you add:
set graph_use_ctrl_key 0
in your xschemrc and press ctrl-m in a graph it does exactly that.
`

https://github.com/user-attachments/assets/0a47a31e-957f-4215-bf77-df67ffc3b5b6

`

StefanSchippers, 2025-03-09:
The **graph_use_ctrl_key** was introduced because some users Don't want anything special happen when hovering over a plot, and instead want all commands for editing a schematic remain active, (may be they want to draw wires or place components over a graph, who knows). One thing that is irritating is if you are navigating the schematic and using the mouse wheel to zoom/unzoom. When over a graph the zoom will zoom the graph instead of the schematic if **graph_use_ctrl_key** is not enabled.

olisnr, 2025-03-09:
yes You are right. changing the cursors over plots in the case of **graph_use_ctrl_key** is inconsequent...