StefanSchippers/xschem #331: CDL export feature update

sergeiandreyev, 2025-03-11:
Is it possible to implement minor changes to the CDL export functionality, to have the start and end keywords in capitals and remove the `.end` at the end?
i.e. change
```
** sch_path: /home/andreev/tmp/io_inv.sch
.subckt io_inv i vdd vss nq
*.PININFO i:I vdd:B vss:B nq:O
M2 nq i vdd vdd sg13_lv_pmos l=0.13u w=4.41u ng=1 m=1
M1 nq i vss vss sg13_lv_nmos l=0.13u w=3.93u ng=1 m=1
.ends
.end
```
to
```
** sch_path: /home/andreev/tmp/io_inv.sch
.SUBCKT io_inv i vdd vss nq
*.PININFO i:I vdd:B vss:B nq:O
M2 nq i vdd vdd sg13_lv_pmos l=0.13u w=4.41u ng=1 m=1
M1 nq i vss vss sg13_lv_nmos l=0.13u w=3.93u ng=1 m=1
.ENDS
```

StefanSchippers, 2025-03-13:
I have added the menu option:
`Simulation->LVS->Upper case .SUBCKT and .ENDS`.
This can be set with the xschemrc TCL variable `uppercase_subckt`
Also if lvs_netlist is enabled the **.end** will not be printed. I think it will not cause problems in CDL netlists.
If someone complains I will add another menu option to enable or disable the final **.end** in netlists.

StefanSchippers, 2025-03-14:
Closing issue. If you experience problems feel free to reopen...

closed 2025-03-14 by StefanSchippers