StefanSchippers/xschem #332: `xschem line` and `xschem polygon` ignore infix_interface

itsayellow, 2025-03-11:
I want to implement a keystroke using `bind .drw` to draw polygons and lines.  I want to use my own tcl function, so I'm using the tcl functions `xschem line` and `xschem polygon`.

I also have `infix_interface` set to 1.

I notice that my keystroke binds work, however I'm not getting the infix behavior when using `xschem line` and `xschem polygon`.  Those tcl commands always require another click to start the line or polygon.

Is there any argument to these xschem functions that would allow their behavior to be infix based on `infix_interface`?  Or some other workaround?

StefanSchippers, 2025-03-12:
Xschem now has these additional commands:

xschem wire gui
xschem line gui
xschem polygon gui
xschem rect gui

That start the related shape placement, using the mouse position as 1st points. These should be used in event bindings.

You can do a quick test this way in the xschem console:

after 2000 {xschem line gui}

you have 2 seconds to go in the main window and set the mouse somewhere. It then will start a line placement.

itsayellow, 2025-03-12:
Same as always: thanks for such helpful info and work!  In particular the `bind [xschem get current_win_path]` was a valuable tip!

I'll try this out soon. 👍

itsayellow, 2025-03-13:
> Same as always: thanks for such helpful info and work! In particular the `bind [xschem get current_win_path]` was a valuable tip!
> 
> I'll try this out soon. 👍

This brings up another issue.  I find my new keybindings don't work for new windows, because I execute the `bind` commands in xschemrc.

Is there a way to trigger my `bind` commands to set the right keystrokes for any new window?  Basically I need to run TCL code when a new window opens.

StefanSchippers, 2025-03-13:
Are you using the tab interface (new tabs are created) or multi-window interface (new windows are created)?

itsayellow, 2025-03-13:
I'm using multi-window interface.

> For multi-window interface I need to provide some mechanism to provide a user procedure to execute at any new window creation.

OK, understood.

I could also use some sort of hook or trace or something when a new design is loaded into the window (through load, hierarchy traversal, etc.).  Does one exist?

StefanSchippers, 2025-03-14:
I have tested and looks like the `user_startup_commands` is executed for any window, including the 1st one.

https://github.com/user-attachments/assets/1b676c70-94f4-4b23-ae6c-53ac923b9027

StefanSchippers, 2025-03-14:
I have tested and looks the commands are executed also for first window.

https://github.com/user-attachments/assets/868ddd9c-4b53-4dc4-a21c-21c6891ef51d

itsayellow, 2025-03-14:
I think there might be an important difference between your example and mine.

I set `user_startup_commands` in a tcl file that is added to `tcl_files`.  It gets executed, but it is executed after my `xschemrc` finishes being sourced.

I'm guessing the new window is created sometime after sourcing `xschemrc` and before sourcing the TCL files in `tcl_files`.

Currently in my TCL file, to execute `user_startup_commands` in the first window, I need to specifically `eval $user_startup_commands`.  

I have a `puts` statement inside of `user_startup_commands`, so it is easy to see when it is and isn't executing.

If it's too awkward to change the current behavior and sequencing, my workaround of `eval $user_startup_commands` is fine.  But it might deserve a note in the `xschemrc` comments or somewhere.

I also tested, and putting `user_start_commands` inside of `xschemrc` works for me too.

StefanSchippers, 2025-03-14:
Ok, it is now fixed. The eval of user_startup_commands is now done *after* executing the tcl_files scripts.
So the user_startup_commands block is executed for all windows.

A suggestion. If you use tk specific instructions (GUI)  protect the block with a if {[info exists has_x]} { ...}:
This avoids error messages if you start xschem in console-only mode (xschem -x)  or on system without an X server.
```
set user_startup_commands {
   if {[info exists has_x]} {
     puts "Adding custom bindings"
     bind [xschem get current_win_path] <Key-plus>  {
       puts Hello
     }
   }
}
```

itsayellow, 2025-03-14:
Yes this is great.  I only need to define my TCL procedures once, but the `bind` commands can easily be added to `user_startup_commands`.  
It's nice to be able to have the `bind` statements in the same *.tcl file as the procedures.

itsayellow, 2025-03-14:
In this case with `user_startup_commands`, don't they only execute on the creation of a window, so isn't `has_x=1` implicit?

StefanSchippers, 2025-03-14:
I have checked. if xschem is started without graphics (-x option or no DISPLAY variable set or no Xserver running) the **user_startup_commands** is still evaluated.

Of course it is evaluated only once since there is no way to open other windows or tabs if there are no windows / tabs.

The approach is debatable. You are not forced to put tk-specific GUI commands (bind and similar) there, you can also put normal tcl commands, perhaps together with toolkit commands.

Using the **has_x** variable lets you, the user decide what to execute if there are windows and what to execute anyway.