StefanSchippers/xschem #334: tcl `xschem net_label ...` cannot find label symbols

itsayellow, 2025-03-13:
I'm not sure if this is a regression (I thought it used to work).  I'm using commit 2a4ebe4b .

`xschem net_label ...` seems to be unable to find the label symbol for pin or wire.

if the full path to the directory containing `lab_wire.sym` and `lab_pin.sym` is in `XSCHEM_LIBRARY_PATH`:
`xschem net_label 0` or `xschem net_label 1` always place a "--MISSING SYMBOL-- lab_{wire,pin}" 

if the full path to the directory containing `lab_wire.sym` and `lab_pin.sym` is **NOT** in `XSCHEM_LIBRARY_PATH`:
`xschem net_label 0` or `xschem net_label 1` always place a "--MISSING SYMBOL-- lab_{wire,pin}" 
and also returns the warning/error in the tcl console: `l_s_d(): Symbol not found: <xschem current working directory>/devices/lab_{wire,pin}.sym`

StefanSchippers, 2025-03-13:
Try to do in the xschem console:

`abs_sym_path lab_wire.sym`
or
`abs_sym_path devices/lab_wire.sym`

Does either of the two commands return the correct path containing the symbol?

I have tried multiple project setups and to me it works.

check if XSCHEM_LIBRARY_PATH items are all correct, with the right `:` separator, if all directories exists and you have permissions to read.

There is also a function in xschem to locate xschem files, using XSCHEM_LIBRARY_PATH:
```
xschem [~] find_file_first lab_pin.sym
/home/schippes/xschem-repo/trunk/xschem_library/devices/lab_pin.sym
```
(There is also a `find_file` functions that returns all files (if more than 1 found))

There is also a match_file function that takes a regular expression as argument:
```
xschem [~] match_file {lab_(pin|wire)\.sym} {} 1
/home/schippes/xschem-repo/trunk/xschem_library/devices/lab_wire.sym /home/schippes/xschem-repo/trunk/xschem_library/devices/lab_pin.sym
```
In the above I have set the 2nd argument to empty. This is a list of paths to search. If empty uses XSCHEM_LIBRARY_PATH. The 3rd argument tells how many levels down to look for the files.

itsayellow, 2025-03-13:
Yes, I have the xschem library in `XSCHEM_LIBRARY_PATH` as `${XSCHEM_SHAREDIR}/xschem_library`

Also abs_sym_path works with `devices/lab_wire.sym` .  I set the directory above devices in `XSCHEM_LIBRARY_PATH`.  In fact my workaround to do the equivalent is the command: `xschem place_symbol [abs_sym_path "devices/lab_wire.sym"]` and this works.

```tcl
xschem [~/work-xschem/nmos_sweep] l_s_d(): Symbol not found: /home/caduser/work-xschem/nmos_sweep/devices/lab_wire.sym

xschem [~/work-xschem/nmos_sweep] abs_sym_path devices/lab_wire.sym
/usr/local/share/xschem/xschem_library/devices/lab_wire.sym
xschem [~/work-xschem/nmos_sweep] puts "$XSCHEM_LIBRARY_PATH"
/usr/local/share/xschem/xschem_library:/home/caduser/.xschem/libs:/home/caduser/git/pdk_xschem/xschem/libs
xschem [~/work-xschem/nmos_sweep] 
```

itsayellow, 2025-03-13:
For some reason it always tries to find `lab_wire.sym` in my working directory.  Or at least that's what the error message refers to.

itsayellow, 2025-03-13:
`find_file_first` also works.

```tcl
xschem [~/work-xschem/nmos_sweep] find_file_first lab_wire.sym
/usr/local/share/xschem/xschem_library/devices/lab_wire.sym
xschem [~/work-xschem/nmos_sweep] find_file_first lab_pin.sym 
/usr/local/share/xschem/xschem_library/devices/lab_pin.sym
``

StefanSchippers, 2025-03-13:
when a component is searched for in the working directory it means all searches failed and component was not found. However since you told that abs_sym_path and find_file works this may be a scope problem.

the xschem net_label command is executed inside some procedure so some global data is hidden. 
This is probably an error that needs to be fixed, and I should be able to do it as soon as I can replicate the problem.

**Just to be sure: does** `xschem net_label 1` **fail if executed directly from the xschem console?**

StefanSchippers, 2025-03-13:
Also verify **nolist_libs** variable is empty.
This tcl variable may contain a list of patterns (usually directory names) you don't want to search for

itsayellow, 2025-03-13:
`nolist_libs` is empty.

```tcl
xschem [~/work-xschem/nmos_sweep] puts "$nolist_libs"

xschem [~/work-xschem/nmos_sweep] 
```

`xschem net_label 1` fails from xschem console.  Or at least it starts placing is an empty box with "MISSING SYMBOL" inside of it.

I did a `xschem debug 1` starting with me executing `xschem net_label 0`:
[xschem_debug_1.txt](https://github.com/user-attachments/files/19235557/xschem_debug_1.txt)

Of note is that `place_symbol` seems to find `devices/lab_wire.sym` initially.
Then `match_symbol` thinks it's found `lab_wire.sym` in `<my_work_dir>/devices/lab_wire.sym`.  Neither directory `devices` nor `devices/lab_wire.sym` exist in that directory. 
```
match_symbol(): name=/home/caduser/work-xschem/nmos_sweep/devices/lab_wire.sym
match_symbol(): found matching symbol:/home/caduser/work-xschem/nmos_sweep/devices/lab_wire.sym
```

itsayellow, 2025-03-13:
At this point I should probably update xschem to the latest code just to make sure.  I'll start doing that.

StefanSchippers, 2025-03-13:
Do a search in the filesystem to see if there are multiple lab_wire.sym files around.

StefanSchippers, 2025-03-13:
Can you provide a directory structure for:

- your working directory. 
- The xschem installation (is it in /usr/local? like there are /usr/local/bin/xschen, /usr/local/share/xschem and /usr/local/share/doc/xschem directories?)
- your XSCHEM_LIBRARY_PATH.

You can also inspect the **pathlist** variable in the xschem console:

`foreach i $pathlist {puts $i}`

**pathlist** is a copy of **XSCHEM_LIBRARY_PATH**, formatted as a valid TCL list and with non existing directories removed.
check if all directories are OK.

StefanSchippers, 2025-03-13:
If everything fails you can also try to roll back to an older xschem release and see if this is a regression or not.

itsayellow, 2025-03-13:
I tried the latest commit e789988b and still the issue occurs (not fixed).

working directory: `~/work-xschem/nmos_sweep`
```shell
docker-eda:nmos_sweep ls
nmos_sweep.spice  simulations/  xschemrc
docker-eda:nmos_sweep 
```

I installed xschem with default locations.  xschem is in `/usr/local/share/xschem`, `/usr/local/bin/xschem`, `/usr/local/share/xschem`

```
xschem [~/work-xschem/nmos_sweep] puts $XSCHEM_LIBRARY_PATH
/usr/local/share/xschem/xschem_library:/home/caduser/.xschem/libs:/home/caduser/git/pdk_xschem/xschem/libs
xschem [~/work-xschem/nmos_sweep] 
```

```
xschem [~/work-xschem/nmos_sweep] foreach i $pathlist {puts $i}
/usr/local/share/xschem/xschem_library
/home/caduser/.xschem/libs
/home/caduser/git/pdk_xschem/xschem/libs
xschem [~/work-xschem/nmos_sweep] 
```

itsayellow, 2025-03-13:
I just tried xschem at commit e9c927d8 and it seems to completely work.

I see no problems with `xschem net_label 0` from xschem console or within my tcl functions.

itsayellow, 2025-03-13:
commit 279d24288dda18715a3ce31f959f4ac01ce80edf (2025-02-14) is the regression.
commits before it work fine, and it and later commits fail

I did a manual binary search of trying commits.

StefanSchippers, 2025-03-13:
Thank you. Can you please attach the schematic file that is loaded in xschem when you want to add the net label?

StefanSchippers, 2025-03-13:
I have removed the offending lines that caused the regression. Can you please test with most recent ?

StefanSchippers, 2025-03-13:
Thank you for the binary search!, it's a tedious operation.

itsayellow, 2025-03-13:
It works now with latest commit 9cb435c4 !  Thanks!

I sanitized some names, but it's basically the same file:
[nmos_sweep_tb.sch.txt](https://github.com/user-attachments/files/19237843/nmos_sweep_tb.sch.txt)

closed 2025-03-15 by StefanSchippers