StefanSchippers/xschem #336: Bug Report

TheSUPERCD, 2025-03-16:
## Bug description

Latest set of changes to xschem introduces a major bug that occurs when drawing wire with `orthogonal_wiring` mode enabled. Trying to place down a wire in `orthogonal_wiring` mode causes the drawn wire to disappear, effectively blocking the user from drawing wires in this mode - completely breaking the existing `orthogonal_wiring` functionality of xschem as shown in the video below:

https://github.com/user-attachments/assets/40e340f0-3100-4f92-9f25-d597b7e3d4cb

## Reason for it's appearance

As far as my understanding goes, making the `drawtemp_manhattanline()` function use local variables `nl_xx1`, `nl_xx2` etc. caused the `xctx->nl_xx1` variables to remain unchanged when the user is drawing the temporary wires in rubber-band-mode in the `new_wire()` function (in `src/actions.c`, since `drawtemp_manhattanlines()` is used there). But when the time comes to place down the wire, the `new_wire()` function checks for the values of `xctx->nl_xx1` variables, since they are supposed to be changed during the rubber-mode templine drawing process.

## Potential fix

This seems to be fixable by replacing the checks for `xctx->nl_xx1` values to use `xctx->nl_x1` values instead (since at this point, the wires are going to be placed down anyway). But since I haven't run a lot of tests to verify that it doesn't break something in a different place, I wouldn't trust this solution to be the perfect fix for this issue yet.

TheSUPERCD, 2025-03-16:
The potential fix has been applied in commit 9d7acdd in my fork. I'll revert and apply the official patch if a different solution is used to address this issue.

StefanSchippers, 2025-03-16:
Yes, I guessed something was impacted by this. 
On the other hand the xctx->nl_[xx|yy][12] used in select_rect and in drawtemp_manhattanline() caused another set of problems with selection of objects. These variables are used for the selection rectangle *and* for the manhattan drawing mode.

StefanSchippers, 2025-03-16:
Applied your fix also in my master branch. Thank you!.

StefanSchippers, 2025-03-16:
I have also appled your changes to my repo. They are now almost identical. There are only a couple of differences in 

- callback.c line 3747 in my repo.
- callback.c line 4509 in my repo.

These changes prevent some graphic artifacts when no cadence_compat is active.

If you believe these can be imported in your branch without breaking something the files will be equivalent.

TheSUPERCD, 2025-03-16:
Thanks a lot for adding the changes! As for the differences between the fork and this upstream repo, the two parts you mentioned are okay in the fork (I think). This is why:

- `src/callback.c` (line 3747): This part in the upstream repo blocks the use of ALT-Z keybind for toggling diamond-shaped snap-cursor (which is not bound to any other action), unless cadence_compat is enabled. So I think this part isn't going to create any problem, and also provide users with a useful feature. 

- `src/callback.c` (line 4509): Here, the `redraw` and `start_wire` functions are conditionally used (after checking for cadence_compat) to place down the second half of an orthogonal_wire on a double-click user action. Since `orthogonal_wiring` is already a standard (non-cadence) feature on xschem, providing a way to make it work as expected should work without the user enabling `cadence_compat` (it's like that in the fork). Also, I haven't noticed any bug appearing due to a double-click during the normal operations of xschem without checking the state of cadence_compat.

So, since these changes don't seem to negatively affect anything, if possible, I would like to suggest for these parts to be adopted in your repo too. Also, I found an old C89 violation at line 119-120 that has been fixed in my fork after you pointed it out in #292. It seems to have crept in with commit 3adcf40 (`src/callback.c` line-116).

I also just found out that the functionality in line 4509 is a little broken - it doesn't work for an orthogonal wire with manhattan_line==2. It was my mistake for not checking this properly. I'll try to get that fixed as soon as possible (mentioned in #297). 

Thank you!

TheSUPERCD, 2025-03-16:
Referencing the previous comment, this is where I fixed the issue from 3adcf40 -----> d4b3881.

StefanSchippers, 2025-03-16:
Thank you for the comments.
Now I need to improve the manhattan style move of wires. I want the ending segments to remain (partially) selected, now the wrong segments are selected after a move in some combinations.

https://github.com/user-attachments/assets/954a7b26-3b3d-4495-af32-2da23c353d40

Also an important note. 

Given a wire (or line) from x1,y1 to x2,y2 the coordinates should always be ordered in increasing X values (so x1 always <=x2) or if x1 and x2 are identical in increasing Y values (so y1 <=y2)

This is what the ORDER() macro does. This needs to be called before storing (with storeobject())  the wire.

Ordered coordinates make some operations faster like line clipping to a small view area. (so we don't have to check everytime what is the lowest  coordinate).
data saved in xschem files is guaranteed to be ordered.

This applies to wires and lines.

Rectangles are also ordered, although the RECTORDER() macro is simpler, just get the lowest x and lowest y for first point and highest x,y for the second point.

[image]

A line with coordinates 10, 10, 5, 20  needs to be stored as 5, 20, 10, 10
A line with coordinates 10, 10, 10, 5 needs to be stored as 10, 5, 10, 10

Also consider that the X11 coordinate system has Y values running from top to bottom. (this is the page description convention (at least for English language), where you start writing from top - left and go to bottom-right)

StefanSchippers, 2025-03-16:
@TheSUPERCD I have leveraged your `drawtemp_manhattanline()` function also for new_line().
Moreover:
global `xctx->nl_xx[12]` and `xctx->nl_yy[12]` variables used in `new_wire()`, `new_line()`, `new_rect()` and `select_rect()` transformed to local variables `nl_xx[12]` and `nl_yy[12]` since these are used as temporary storage and not state variables. This reduces risk of contention (same var used in different places in nested operations).

I hope this does not break too many things. In the mean time will keep doing some tests.

Best regards

TheSUPERCD, 2025-03-17:
Thank you for adding in the changes! As far as I've seen, nothing seems to be broken after the last update. I've also added the ordering that was missing in `move_objects()` after you pointed out the problems with the previous version.

And for the wrong partially selected wire-segments in the video, I think it's because the wire-extension process uses `start_wire()` function, which adds in a new wire segment for orthogonal_wiring, and the selection stays with the old segment. A potential fix would be to swap the selection field for the parent and child segments (if the child exists).

StefanSchippers, 2025-03-18:
https://github.com/user-attachments/assets/f2420daf-2b2f-42c9-beac-c88f53e98e7d

TheSUPERCD, 2025-03-18:
Hi, sorry for the unusually late response. I had been feeling sick all day today and hadn't had the chance to properly look at the problem. 

The reason for the problem seemed to be a single `manhattan_lines` value being forced on all connected wires, which is dynamically assigned in the `drawtemp_mahattanline()` function. Doing the same here for each independent wire solved the issue. I have added the fix and synced the forked repo with your master branch now. 

Also, thanks for posting the helpful video explanation! It helped out a lot!

StefanSchippers, 2025-03-18:
Great!, Happy my changes have not destroyed everything. I thought more changes were necessary to make orthogonal_wiring work.
Well done and thank you.

TheSUPERCD, 2025-03-18:
There is one last change I have made to this (commit 02f0a4e on my fork). Since the code for reassigning manhattan_lines in orthogonal_wiring-mode appears in multiple places, it is now bundled in a function (`recompute_orthogonal_manhattanlines()` defined in `src/actions.c`). 

I made the old mistake of C89 violation with the last patch. This should take care of that problem once and for all.

closed 2025-03-19 by TheSUPERCD