StefanSchippers/xschem #337: Window scaling

Blebowski, 2025-03-17:
Hi,

I would like to use XSCHEM under RHEL8 with XFCE.
I am running into issuues scaling the window somewhat comfortably.

I searched through the manual and issues, and I found it is possible
to scale the font size (`set tk_scaling XYZ`. But this does not scale the icons.

E.g. without scaling:
[image]

With scaling:
[image]

The menu that pops-out when doing right-click is not scaled either.

Is it possible to implement a global scaling of everything right in XSCHEM ?
This would make it much simpler that trying to google different solution for each
of Gnome, KDE, XFCE.

StefanSchippers, 2025-03-17:
Hi, Ondrej

I will add the option to scale the icons in toolbar and context menus.

A global scaling will not fit every user's taste, since there will be users with 800x600 displays up to XXXUHD displays.

Toolbar and context menu buttons are based on bitmap images. Old technology, but this makes xschem run unchanged on all Tk toolkit libraries released in the last 20 years, as well as the future ones (Tk 9.0).

I think the full xschem UI scaling will be done this way:
```
set tk_scaling 2.0
#default: 24
set toolbar_button_size 48
# default: 16
set ctxmenu_button_size 32
```

I will add in the next days the above options (in red) to rescale toolbar/ctxmenu icons.

Usually desktop environments set the DPI for the monitor, graphical applications (tk / xschem included) rescale fonts and widgets accordingly.
I don't use desktop environments so I don't either know where these setting are stored and applied. One of the many reasons I removed them from my systems.

On Xorg systems all the Freetype (fonts) stuff is scaled and tuned on these .Xresources settings (i copied here my settings, mostly commented):
```
!-- Xft settings -- !
! Xft.dpi: 96
! Xft.core: True
! Xft.render: False
! Xft.scale: 4.0
Xft.antialias:  true
!!! none, rgb, bgr, vrgb or vbgr
Xft.rgba:     rgb
Xft.hinting:    true
!!! hintfull, hintslight
Xft.hintstyle:  hintslight 
```

Gtk widgets seem to be scaled based on these shell variables:
`export GDK_DPI_SCALE=1.5; export GDK_SCALE=1`

Qt on the other hand uses:
`export QT_AUTO_SCREEN_SET_FACTOR=0; export QT_SCALE_FACTOR=1; export QT_FONT_DPI=128`

(Xschem does not use Gtk/Qt).

Usually none of the above settings does what you think. The only way to make it work is to do multiple iterations of change, test, change, test, change, test...

Thank you for pointing out this scaling issue on Hi-res monitors.

Blebowski, 2025-03-18:
Hi @StefanSchippers , what you propose sounds like ideal solution, thank you.

StefanSchippers, 2025-03-18:
@Blebowski I have added two variables you can set in xschemrc file:
```
toolbar_icon_zoom
ctxmenu_icon_zoom
```
these are integers. You can set to 2 to have double icon sizes:
```
set toolbar_icon_zoom 2
set ctxmenu_icon_zoom 2
```
May be in the future better image formats will be used like SVG to allow fractional scaling.
For the time being this is what I can do.

Together with tk_scaling (this is a floating point number) I think you can get a reasonable window scaling.
This is the result on my desktop (which does not have a 4k display):

[image]

Blebowski, 2025-03-18:
Thanks, would you be willing to do a `3.4.7` release please ?

Blebowski, 2025-03-18:
@StefanSchippers thanks, it looks much better now.

Still the text in the right-click pop-out menu is quite small:

[image]

Would it be possible to scale that-one too (via separate variable or by the tk_scaling) ?

Blebowski, 2025-03-18:
Beyond this one thing, it seems good.

Blebowski, 2025-03-18:
The configure output:
```
Looking for a shell... /bin/sh -c
Looking for a shell escape character... found: '\'
Detecting temp dir... validated /tmp/xschem/scconfig/
Checking for system type... OK (name: Linux; class: UNIX)
--- Detecting target (same as host)
Checking for cc (target)... OK ('gcc', ' -pipe', '')
Checking for -fpic... OK (-fPIC)
Checking for popen(3)... OK ('#define _XOPEN_SOURCE\n', '' and '')
Checking for dup2(2)... OK ('', '' and '')
Checking for flex... Found.
Checking for bison... Found.
Checking for XOpenDisplay... OK ('', '' and '-lX11')
Checking for tk... 8.6... OK ('', '' and '-ltcl8.6')
OK ('', '' and '-ltcl8.6  -ltk8.6')
Checking for awk... OK (awk)
Checking for xpm... OK ('', '' and '-lX11 -lXpm')
Checking for cairo... OK ('', '-I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/uuid' and '-lcairo')
Checking for libjpeg... OK ('', '' and '-ljpeg')
Checking for stdint.h... OK ('', '' and '')
Testing size of type             unsigned char...  OK, size 1 byte
Testing size of type        unsigned short int...  OK, size 2 bytes
Testing size of type              unsigned int...  OK, size 4 bytes
Testing size of type         unsigned long int...  OK, size 8 bytes
Testing size of type    unsigned long long int...  OK, size 8 bytes
Testing size of type               signed char...  OK, size 1 byte
Testing size of type          signed short int...  OK, size 2 bytes
Testing size of type                signed int...  OK, size 4 bytes
Testing size of type           signed long int...  OK, size 8 bytes
Testing size of type      signed long long int...  OK, size 8 bytes
Testing size of type                   uint8_t...  OK, size 1 byte
Testing size of type                  uint16_t...  OK, size 2 bytes
Testing size of type                  uint32_t...  OK, size 4 bytes
Testing size of type                  uint64_t...  OK, size 8 bytes
Testing size of type                    int8_t...  OK, size 1 byte
Testing size of type                   int16_t...  OK, size 2 bytes
Testing size of type                   int32_t...  OK, size 4 bytes
Testing size of type                   int64_t...  OK, size 8 bytes
Testing size of type                     float...  OK, size 4 bytes
Testing size of type                    double...  OK, size 8 bytes
Testing size of type               long double...  OK, size 16 bytes
Found best fit 1 bytes wide uint: unsigned char
Found best fit 1 bytes wide sint: signed char
Found best fit 2 bytes wide uint: unsigned short int
Found best fit 2 bytes wide sint: signed short int
Found best fit 4 bytes wide uint: unsigned int
Found best fit 4 bytes wide sint: signed int
Found best fit 8 bytes wide uint: unsigned long int
Found best fit 8 bytes wide sint: signed long int
Found best fit 4 bytes wide float: float
Found best fit 8 bytes wide float: double
Found best fit 16 bytes wide float: long double
Checking for cairo... OK ('', '-I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/uuid' and '-lcairo')
Checking for cairo-xcb... OK ('#include <cairo-xcb.h>\n', '-I/usr/include/cairo -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/uuid' and '-lcairo')
Checking for xcb... OK ('', '' and '-lX11 -lxcb')
Checking for xgetxcbconnection... OK ('', '' and '-lX11 -lX11-xcb')
Checking for xcb_render... OK ('', '' and '-lX11 -lxcb-render')
--- Detecting runtime (same as host)

--- Generating build and config files
config.h:              ok
Makefile.conf:         ok
doc/manpages/xschem.1: ok
src/Makefile:          ok

=====================
Configuration summary
=====================

Compilation:
 CC:        gcc
 debug:     no
 profiling: no

Paths:
 prefix:        /tools/opensource/xschem/2025_03_18_c4b3e20
 user-conf-dir: ~/.xschem
 user-lib-path: ~/.xschem/xschem_library
 sys-lib-path:  /tools/opensource/xschem/2025_03_18_c4b3e20/share/xschem/xschem_library/devices

Libs & features:
 tcl:       -ltcl8.6
 tk:        -ltcl8.6  -ltk8.6
 cairo:     yes
 libjpeg:   yes
 xcb:       yes

Configuration complete, ready to compile.

```

StefanSchippers, 2025-03-18:
I have pushed a small change so that right-click menu use the same default font as all other GUI elements. Hopefully this makes tk_scaling more consistent.
Please test if you have a chance.

Blebowski, 2025-03-18:
Yes, that works, thanks.

Now, I still have one more issue, and that is drawing of the right-click menu. It takes noticeable amount of time sometimes:

https://github.com/user-attachments/assets/bca7eafe-03b8-4fcf-81a4-09b624644525

Blebowski, 2025-03-18:
I am using Linux, RHEL8, X11 (I think).

Blebowski, 2025-03-20:
Thanks for the tips.

My setup was following:

My machine -> X2Go Connection to a "Terminal Server" (for code editing, full screen share)
Within the terminal server -> SSH -X to "Runner" (a machine that has all the libs to run EDA tools, so the ASIC work like simulation, debug, etc. is done on Runner).

Then trying to run the XSchem on the Runer.

Turns out the problem is the SSH connection performance. When I run the tool on the underlying "Terminal Server" there are no performance issues. Strange is, that when we run commercial digital tools (either legacy like DVE) or more modern based on Qt, teh full GUI windows get forwarded OK over SSH and they are workable.

So, I think we can leave this and we stick with avoiding launching XSchem over SSH (unless you have some super tip :) )

StefanSchippers, 2025-03-20:
I have done some tests over ssh, and yes, there is a performance issue when drawing GUI widgets.
I remember ssh was much faster years ago, so may be the developers have fucked it up or pythonized some components.

I have tried connecting both to a fast and a very slow machine and there is no performance difference, so ssh is the bottleneck. 

The right click menu displays exactly as in your video, very slow.

I have tried connecting over a wifi network to a slow Samsung N220 computer with an Atom processor, where a vnc server is running and performance is very good. See video.
There is definitely a problem with SSH.

https://github.com/user-attachments/assets/24660560-7698-4d63-9372-27edc45ca36b

Blebowski, 2025-03-20:
```
I remember ssh was much faster years ago, so may be the developers have fucked it up or pythonized some components.
```
:D :D

I am glad that I am not the only person who is getting annoyed by "yet another completely useless python abstraction which
makes things sexy and pythonistic (whatever that means), but screws them up in basically every other regard..."

My experience is basically same as in your video (including startup issues).

I will try to dig if there are some options to speed it up (just out of curiosity).

I will close this as there is nothing to resolve anymore.

closed 2025-03-20 by Blebowski