StefanSchippers/xschem #338: Deselection thru rectangle broken

ananthchellappa, 2025-03-19:
Bug has been introduced in the last month. My fork doesn't show the issue..

Press 'd' and click on object to deselect works fine.
Pressing SHIFT-d and drawing a rectangle - after 2nd click, everything gets deselected and the rectangle persists and only gets cleared by the fit command ('f').

[image]

StefanSchippers, 2025-03-19:
Thank you. I think it is now fixed. Was indroduced after this commit:
```
commit 936b72c7fc875bcb130ab56f11c38354b8fbcdff
Author: stefan schippers <stefan.schippers@gmail.com>
Date:   Tue Feb 18 11:49:32 2025 +0100

    Escape pressed while drawing polygon will cancel last segment instead of the whole poly

```

ananthchellappa, 2025-03-19:
Seems to be fixed. Is it possible to explain a bit so that we learn something?

As I am fond of saying, the idea of what the product is should change. It's not a schematic editor, but, because others are wiling to contribute, the code for the schematic editor is the product. So we need to improve the code quality. It is currently very difficult to reverse engineer.

StefanSchippers, 2025-03-19:
A button1 press (in handle_button_press() ) starts a new selection operation, but this must not happen if an area selection is already ongoing. This is what the part in bold does, and got removed accidentally during a refactoring. 

if(!excl && **!(xctx->ui_state & STARTSELECT)**) {

The excl variable is:

int excl = xctx->ui_state & (STARTWIRE | STARTRECT | STARTLINE | STARTPOLYGON | STARTARC)

So the button1 click does not start a selection if a wire, rectangle, line, polygon, arc placement is ongoing **and if a selection is ongoing**.

closed 2025-03-19 by StefanSchippers

ananthchellappa, 2025-03-20:
Thanks Stefan!

Going forward, is it possible to use better variable names for new variables? I don't think excl is a good name.