StefanSchippers/xschem #339: Can't read "sim(spice,3,fg)": no such element in array

simonwaid, 2025-03-19:
Within the IHP PDK, I was trying to execute some simulations from the start page using Xyce. In e.b. tran_mim_cap.sch, clicking on SimulateXyce should set Xyce as the simulator and run the simulation. However, the command `set sim(spice,3,cmd) {Xyce \"$N\"} ` Seems to fail with the error message: "Can't read "sim(spice,3,fg)": no such element in array"

StefanSchippers, 2025-03-19:
what is the xschem version you are using?

`xschem --version`

StefanSchippers, 2025-03-20:
May be you have a **simrc** file in the directory where simulations are run. Delete that file if any found.

If still not working go in the src/ directory of xschem, open the **xschem.tcl** file, search for **proc set_sim_defaults**.
Down in the procedure there should be a line:
**set_ne sim(spice,3,fg) 0**
 (**set_ne** means _set if not existing_) 
If that file is present the array element exists and the warning message can not happen.
**set_sim_defaults** is executed when you trigger the launcher.

If still you can't get it working do a global filesystem search for any directory named **xschem** and any executables named **xschem**, delete everything and start over again with a clean xschem installation.

simonwaid, 2025-03-21:
Thank you! Deleting **~/.xschem/simrc** did the trick.

closed 2025-03-21 by simonwaid