StefanSchippers/xschem #340: xctx->semaphore

ananthchellappa, 2025-03-20:
Not a real "issue", just looking for help understanding the code. 

chatGPT says:

Hardcoded Magic Numbers (xctx->semaphore >= 2): Use named constants instead:

#define SEMAPHORE_THRESHOLD 2
if (xctx->semaphore >= SEMAPHORE_THRESHOLD) { ... }

------------

Could you say a bit about what the name means (in the context of Xschem. I know what semaphore means in general) and what the different possible values mean and what -- and ++ to xctx->semaphore mean? (as in, wouldn't it be better to assign the desired value? Why/why-not?)

Thanks!

StefanSchippers, 2025-03-20:
Some actions (like changing something in a schematic) need to lock the UI (so no nested modifications can take place)  the `semaphore` variable is used to prevent this. Some "modify" actions need to be completed before accepting new ones. However some other actions (like zoom or move in a schematic) are allowed even if semaphore is set, so the xschem app remains 'responsive' and not frozen.

Moreover an X event calls the callback() function. Another event dispatched before callback() has completed needs to be handled. Typically the variable is incremented (++) when starting an exclusive operation and decremented when done.

ananthchellappa, 2025-03-20:
Thanks Stefan.

When you say incremented, decremented, what does going from 0 to 1 or 1 to 2 signify? Do the values mean something? Is that captured somewhere in comments?

Thank you

ananthchellappa, 2025-03-20:
For example, here

`if(!excl && button == Button3 && state == ControlMask && xctx->semaphore <2)`

What does semaphore < 2 mean?

StefanSchippers, 2025-03-20:
when the callback() function is entered the semaphore is incremented.
A value f semaphore==1 is normal event servicing.
if another call to callback is done when previous one is not completed 
semaphore is further incremented and goes to 2.
Some commands are not accepted in this case (UI locking).

One example.

- Press 'q' to edit a symbol attribute (do not close the dialog box)
  - **move the mouse**
  - **rotate the mouse wheel**
  - **press arrow keys to move the schematic**
  - press Insert to insert a component
The commands in bold are allowed, the last one is not allowed until the edit attribute dialog box is closed.

StefanSchippers, 2025-03-21:
semaphore can be incremented multiple times, depending how the user is able to nest different read-only commands (commands that do not modify the schematic, and are allowed to nest one into the other).
The key point is that any increment is followed by a decrement when event handling is done.

Also in some scripts one might increment the semaphore value to prevent changes in the schematic until the script has completed. This avoids race condition (script processing and changing a schematic and user also trying to change a schematic).

closed 2025-03-25 by StefanSchippers