StefanSchippers/xschem #341: Selected.type

ananthchellappa, 2025-03-20:
Apologies again for raising an issue for a question about the code.

`typedef struct
{
  unsigned short type;
  int n;
  unsigned int col;
} Selected;`

What does "type" mean? Isn't it better to have an enumerated data type so that we can use meaningful labels for the various possible values?

In callback.c, we check:

`  if(!excl && button == Button3 && state == ControlMask && xctx->semaphore <2)
  {
    Selected sel;
    sel = select_object(xctx->mousex, xctx->mousey, SELECTED, 0, NULL);
    if(**sel.type**) select_connected_nets(1);
  }`

What does type being non-zero mean? Is that documented somewhere?

Thanks!

ananthchellappa, 2025-03-20:
OK, I see, in xschem.h:

```
#define WIRE 1              /*  types of defined objects */
#define xRECT  2
#define LINE 4
#define ELEMENT 8
#define xTEXT 16
#define POLYGON 32
#define ARC 64
```

So, we should probably use

```
typedef enum {
  WIRE = 1,
  xRECT = 2,
  LINE = 4,
  ELEMENT = 8,
  xTEXT = 16,
  POLYGON = 32,
  ARC = 64
} OBJ_TYPE;
```

And

```
typedef struct
{
  OBJ_TYPE type;
  int n;
  unsigned int col;
} Selected;
```

Or something like that? Thoughts?

StefanSchippers, 2025-03-20:
I think it's just minor syntactical sugar.
Whether a macro or an enum label the same code will be generated.
An enum is nicer, may be worth switching if the effort is limited.

StefanSchippers, 2025-03-21:
> In callback.c, we check:
> 
>   {
>     Selected sel;
>     sel = select_object(xctx->mousex, xctx->mousey, SELECTED, 0, NULL);
>     if(sel.type) select_connected_nets(1);
>   }
> 
> What does type being non-zero mean? Is that documented somewhere?

If the ctrl-right mouse button is pressed on empty area the select_object() function returns an empty sel structure, and sel.type is set to 0. Checking for sel.type==0 avoids calling the select_connected_nets() function.

closed 2025-03-21 by ananthchellappa