StefanSchippers/xschem #344: Clicking on a selected instance doesn't change selected set

ananthchellappa, 2025-03-22:
Expected : multiple objects selected. Now, click on an object (wire, instance, etc) (not SHIFT LMB click, but just an LMB click). We expect ONLY this object to now be selected. That is how it is in Cadence

Actual : Nothing changes - all selected objects are still selected.

Is this intentional? Is it easy to tie to a variable so we can have a Cadence-compatible version?

StefanSchippers, 2025-03-22:
if you click an object that is not selected it will become the only one selected, all others will unselect.
If you click on an object that is already selected nothing happend.

ananthchellappa, 2025-03-22:
Thank you Stefan :) If you don't mind, could you answer the other questions also?

StefanSchippers, 2025-03-22:
Yes I think the other behavior can easily be tied to cadence_compat variable.

ananthchellappa, 2025-03-23:
Thanks Stefan. Could you give me a clue as to where to start? Is it entirely within select_object?

StefanSchippers, 2025-03-23:
It is already done. Please test
```
commit d765f6ac4f123ccdab526910e4fcc0f6f37fe6de
Author: stefan schippers <stefan.schippers@gmail.com>
Date:   Sun Mar 23 00:25:29 2025 +0100

    in cadence_compat mode, when a selection is present, clicking on a selected item will unselect all others (default is do nothing)
```

closed 2025-03-23 by ananthchellappa