StefanSchippers/xschem #345: How to add a new function supported by the TCL evaluator?

ananthchellappa, 2025-03-22:
Why? this could be one way to achieve the goals of logging all interaction and key-binding customization.

Currently, we use switch or if statements to go through various key combinations in the event handler.

We check for a key combination and run appropriate C code.

But, if we can capture the C code in a function that can be run through the TCL evaluator, then we just publish the name of the function and let the user bind it to a key combination.

The flexible key bindings can be done with a table that contains the key names (CTRL, ALT, SHIFT, etc) and the TCL function - one per row of the table.

For example, in handle_button_press:

```
static void handle_button_press(int event, int state, int rstate, KeySym key, int button, int mx, int my,
                                double c_snap, int draw_xhair, int crosshair_size, int enable_stretch, int aux)
{
  int use_cursor_for_sel = tclgetintvar("use_cursor_for_selection");
  int START_W_A_L_P_R = xctx->ui_state & (STARTWIRE | STARTRECT | STARTLINE | STARTPOLYGON | STARTARC);
  dbg(1, "callback(): ButtonPress  ui_state=%d state=%d\n",xctx->ui_state,state);
  if(waves_selected(event, key, state, button)) {
    waves_callback(event, mx, my, key, button, aux, state);
    return;
  }

  if( check_and_handle_sch_pan_case() ) return; // refactoring suggested by chatGPT

  /* select instance and connected nets stopping at wire junctions */
  if( non_start_WALPR_and_CTRL_RMB_and_sem_LT_2(START_W_A_L_P_R, button, state) ) return;

....
```

where we have

```
int non_start_WALPR_and_CTRL_RMB_and_sem_LT_2(int START_W_A_L_P_R, int button, int state)
{
  if(!START_W_A_L_P_R && button == Button3 && state == ControlMask && xctx->semaphore <2)
  {
    Selected sel;
    sel = select_object(xctx->mousex, xctx->mousey, SELECTED, 0, NULL);
    if(sel.type) select_connected_nets(1);
    return 1;
  }
  else return 0;
}
```

But, it would be nice to have a function "sel_inst_nets" that lets us say:

if __conditions__ then tcleval("sel_inst_nets 1") // pseudocode

If there is something I can look at (read) to see how to collect C code into a new function supported by tcleval, please share.. 

Thanks

StefanSchippers, 2025-03-22:
events are already handled by TCL (xschem was initially developed as pure C code, with no tcl and no toolkits). Look for `proc set_bindings` in xschem.tcl.

this function now does only a gross-binding (Enter, Motion, Expose, Configure, KeyPress, ButtonPress and so on) and calls C function `callback()`, essentially forwarding all the event details (event type, key, mouse button, mouse coordinates, modifier keys, the window where event originated from (xschem handles multiple windows), and so on.

Using some indirection layer to make the whole thing customizeable by users is probably a lot of work, and we also must prevent bug submissions from users who first screw up the whole interface and then complain xschem is not working.

ananthchellappa, 2025-03-23:
Yes, I agree, but I feel we can get to the "perfect" version eventually. For now, I want to get to the customizability by calling tcleval for each key-combination detected. 

Could you tell me the newest function added to be supported by tcleval? Maybe I can look there and get a start..