StefanSchippers/xschem #347: A way to decouple stretch and selection of attached wires

ananthchellappa, 2025-03-25:
In Cadence, the default is to stretch - that is, if you click on an instance and drag it, the wires stay attached and are redrawn appropriately.
There is no such thing as selecting attached wires with a single click. There is a way to "propagate" the selection - select more attached wires.

I bring this up because, if, in Xschem, the user "enables stretch" ('y' shortcut), then, the user gets this "stretch" behavior but the side-effect is that clicking on an instance (or other object such as a wire) also selects attached wires - which is not desired.

Is there a way to decouple these - that is, keep the "stretch" behavior - move with wires attached, but not select more than you wish to when you click on an object?

Thanks

[image]

ananthchellappa, 2025-03-25:
What I notice is that, even in enable_stretch mode, if you use the selection rectangle in "left" mode, then it is possible to select just the instance, so I think it shouldn't be too hard to to the "decoupling" that I am interested in.. - which could be useful to users who might want to copy an instance without wires.. 

[image]

StefanSchippers, 2025-03-25:
you probably need to set this:

#### if set to 1 at end of a move operation that stretches wires attached to
#### moved objects these wires will be unselected.
#### default: not enabled (0)
set unselect_partial_sel_wires 1

ananthchellappa, 2025-03-25:
I tried that Stefan... didn't work

StefanSchippers, 2025-03-26:
https://github.com/user-attachments/assets/d1afeedf-d1d4-45f8-bbec-36b7bb5431ea

ananthchellappa, 2025-03-26:
Not sure what is being demonstrated 😊 you seem to get the same thing I am seeing..

StefanSchippers, 2025-03-26:
After thwe move wires are no more selected.

ananthchellappa, 2025-03-26:
😊 Thanks Stefan

Is that what I'm looking for? :)

closed 2025-03-26 by StefanSchippers

StefanSchippers, 2025-03-26:
I don't understand your problem. Please explain in detail.

ananthchellappa, 2025-03-26:
When I click on an object (wire, instance, etc), I want to select only that object, not any wires attached to it.

Noting to do with what happens AFTER moving, etc..

StefanSchippers, 2025-03-26:
I don't understand what is the problem. 
because the wires get colored in grey? This is a feature, not a problem, so you immediately  realize you are in stretch mode and the attached wires will move. After the move you can decide if you want to leave these wires partially selected or not (using unselect_partial_sel_wires).

ananthchellappa, 2025-03-26:
🙂

The problem is that I don't want anything other than the instance to be selected.

There is no problem like this in Cadence - that you need to ALSO select the attached wires to move the instance in a user-friendly way..

All I am looking for is a way to have the stretch behavior by default (i.e., enable_stretch = 1) AND be able to select an instance by itself (which I can do by bending over backwards - that is, leftwards click and drag 🙂)

I understand it's a heroic effort to build this tool that people are using..

StefanSchippers, 2025-03-26:
I don't know what Cadence does. There are many things of Cadence I don't like at all. Not showing wires that are about to stretch is one of the many things. Showing in grey the attached wires in stretch mode is deliberate.
I can not change that, since existing users are used to this. You can bind this difference with the cadence_compat variant.

ananthchellappa, 2025-03-26:
Yes Stefan,
        It can be done - I am asking how 🙂

That was the original question - "is there a way" 🙂

Currently, as I said, we can bend over backwards and use leftwards selection with a rectangle, but that is not user-friendly..

I have used Cadence almost 30 years and never had a complaint about the UI 🙂 while using it, I put in about 700 tickets with Cadence - for features and bug-reports and how-to-questions.. There's a reason it's the #1 tool 🙂 I have even changed jobs just to get back to Cadence - you spend so much of your day with the tool that UX is critical to be happy 🙂

What I am trying to do is get Xschem to be as good. Just saying I don't like some things about it doesn't help 🙂

Yes, I understand you have too much to work on. You can use help..

ananthchellappa, 2025-03-26:
BTW, I didn't get an answer to the motivational question 🙂

What motivates you to work on Xschem?

StefanSchippers, 2025-03-26:
if stretch mode is enabled (y key) you can still select an instance alone (no wire stretching) by using ctrl key

similarly if stretch is Not enabled you can stretch-move an instance by clicking with control key

StefanSchippers, 2025-03-26:
The problem with Cadence is that it is dog slow and too big. Also the look (expecially fonts) is terrible.
my motivation for xschem is that I needed a faster simulation loop: edit-netlist-simulate.
For any medium sized project 100ms is the maximum accepted netlist time on a slow computer. In corporate setups I have worked with, Cadence takes 20+ seconds to spit out a netlist on a fast computer. After the netlist was generated from cadence everything else (testbench creation, stimuli creation, run simulator) was done outside of it to save time. This was the approach every designer converged to after some time.

So in all my projects in the last 15 years all schematics and simulations were done in xschem, and schematic was drawn into Cadence only at the very end with final signoff simulation to comply with corporate rules.

ananthchellappa, 2025-03-26:
Hi Stefan,
         If I have CTRL pressed when I click on an instance or wire, I get this :

[cid:5b093f5d-7c47-40d4-9519-a65f7b8e64c9]

StefanSchippers, 2025-03-26:
Remember, when using ctrl to click an object you also must move it somewhere, otherwise you trigger the launcher (on button release, if no movement has been done).

ananthchellappa, 2025-03-26:
🙂

So now you start to understand my problem 🙂