StefanSchippers/xschem #348: Working with multiple windows, sometimes original window stops accepting bind key input

itsayellow, 2025-03-27:
Sometimes when I am working with multiple windows, when I go back to the original window (.drw) and focus it, it ignores all keystrokes.  It seems to only happen when I open one or more other windows.  Closing other windows doesn't help.  Focusing back and forth between other windows doesn't help.  In this state I must close xschem and restart in order to get main window keystrokes working again, as far as I know.

I wish I could be more specific here, but I don't exactly know what conditions cause this bug to occur.  I thought I'd at least make everyone aware, and will post more info here if I have more understanding.

itsayellow, 2025-03-27:
This doesn't happen all the time.

itsayellow, 2025-03-27:
Hm, this just happened on another window than the original window.  (e.g. .x1.drw).  So I guess it's not limited to the original window.

StefanSchippers, 2025-03-27:
Ok, thank you for reporting. 

Handling multiple windows within one single process is one of the hardest things to get done in teh multi-dimensional space of window managers, focus models, Xservers, RDPs.

I need to get this sorted out once and forever of revert back to the old school (separate processes, one per window).

StefanSchippers, 2025-03-27:
> Sometimes when I am working with multiple windows, when I go back to the original window (.drw) and focus it, it ignores all keystrokes.

Does it also ignore mouse actions?

StefanSchippers, 2025-03-28:
I have done some cleanups in the event delivery code. Removed some redundant context switching events.
I don't know if the issue is resolved. At least the code is now simpler (removed accumulated patches and spaghetti-code)

itsayellow, 2025-03-30:
This time a window stopped taking keyboard input, and afterward I did a mouse drag select, and it select "phantom objects" that don't exist or exist offset from real objects.  The images below are just during me selecting a box with a mouse, they are NOT during a move.

[image]

[image]

StefanSchippers, 2025-03-30:
I think the original problem (no keyboard focus) must be solved, just like in case of complier errors you must look at and fix only the first error.

Will try all weird combinations to get an unresponsive window. Problems that take so long to fix exist because I cant easily reproduce them.

StefanSchippers, 2025-03-30:
When you get again such an unresponsive window you can try these commands in the xschem command prompt, while the window has had the last focus:

- `xschem get semaphore`
- `xschem get current_win_path`
-  `xschem switch .drw` or `xschem switch .x1.drw`

The first command must return 0. If it is not 0 the user interface is fucked up. `semaphore` is a mutex variable that is set while servicing an event to avoid a nested 'write' event to occur. By 'write' I mean events that change the schematic data. When event servicing is finished this variable must return to 0.
If you see non zero values reported please report here.

The second command returns the window xschem 'thinks' is focused. .drw is the first window, .x1.drw, .x2.drw, ... are the additional windows. If you know which window had the last focus (click in the window, usually)  and the returned value is different please report.

The last command is a programmatic way to explicitly do a context switch to the indicated window
You can try to use this command to switch to the malfunctioning window and see if it resumes.

You can also use:
`- xschem set semaphore 0`

to reset the UI mutex to 0, this is not a solution to the problem, but can be used to see if window resumes (if this variable was not 0).

itsayellow, 2025-03-31:
Thanks that'll help with debugging.

itsayellow, 2025-03-31:
1. I opened xschem with a design
2. I opened a new window and inside it I used menu File->Open to try open a new design
3. Instead of opening the design in the second window (where I was using the menu) the design opened in the first window.
4. Then Both windows had no keyboard functionality

In the prompt:
```
xschem [~/work-asap7/imult_4bx4b] xschem get semaphore
2
xschem [~/work-asap7/imult_4bx4b] xschem get current_win_path
.drw
xschem [~/work-asap7/imult_4bx4b] xschem set semaphore 0
```

After executing the last command, both windows started working again.

StefanSchippers, 2025-03-31:
Thank you for the info.

for the open file are you using the new symbol / file browser (`set new_file_browser 1` in xschemrc)  or the legacy one (`set new_file_browser 0` or not set at all)?

StefanSchippers, 2025-03-31:
https://github.com/user-attachments/assets/63d14e3d-f8db-4df0-a016-92a01648aeb8

itsayellow, 2025-03-31:
> Thank you for the info.
> 
> for the open file are you using the new symbol / file browser (`set new_file_browser 1` in xschemrc) or the legacy one (`set new_file_browser 0` or not set at all)?

Not set at all.

itsayellow, 2025-03-31:
I submitted the comment https://github.com/StefanSchippers/xschem/issues/348#issuecomment-2764969086 because it was the simplest example that resulted in something not working right.  But unfortunately I can't replicate it now either.

I'm using XQuartz, in click to focus mode.  Also using the option "click in inactive window focuses window AND sends the click to the window."

itsayellow, 2025-04-01:
Just happened when only one window open. 

I started wiring with the "w" key, but couldn't stop because it wouldn't hear my "Esc" key.

itsayellow, 2025-04-01:
I'm using commit 06632ffb now.

StefanSchippers, 2025-04-02:
Try to update to latest, so we are on the same codebase.
If you can reproduce again the problem with the single window:
get the mutex value:
`xschem get semaphore`
if it is not zero edit `callback.c`, set the line `4699`:
from
`#if 0`
to
`#if 1`

rebuild xschem, 
start xschem with: `xschem -l logfile`
try to reproduce the problem
if you can send me the `logfile`.
this file contains all events delivered to xschem.

itsayellow, 2025-04-03:
I think the time I saw the problem with only a single window open (https://github.com/StefanSchippers/xschem/issues/348#issuecomment-2767858862) was after I had multiple windows opened and closed.  For a while afterwards things were fine with one window, but then the error situation happened.