StefanSchippers/xschem #349: Inconsistency in display during partial (leftward) deselection operation

ananthchellappa, 2025-03-28:
Not serious, but worth recording as it would be nice to get it to behave consistently.

In selection - click and drag - move leftward for including objects crossed by the selection rectangle. Observe : as an object is crossed, it is selected and the display indicates this.

In deselection - use ALT-click-and-drag or SHIFT-D - move leftward for deselecting objects that are crossed by the selection rectangle. Observe : though the final selected set is correct, the display is not consistent - the display indicates deselection only after the object is enclosed (rather than crossed) by the deselection rubber-band rectangle.

If fixing, it would be nice to add a variable with this effect:

selection_rect_non_retracting (default 1) meaning: if users sets to 1, then only the final state of the selection rectangle decides which objects are added/removed to/from selected set. 

If adding this, it would be nice if the display updates dynamically as the user changes the selection rectangle. That is, currently, once an object has been selected, or deselected, there is no going back - the user cannot shrink the selection rectangle to undo selection or deselection of an object. But, if this setting is provided, she will be able to, and the display will reflect that an object has been removed from or added to selected set as the rectangle is changed.

[image]

ananthchellappa, 2025-03-28:
Here, I'm not sure if this is expected or not, but it does seem inconsistent to me.

Observation : with enable_stretch=1, when you select rightwards (i.e., not in cross-to-select mode), if you enclose the pin of an instance with the selection rectangle, the wire attached to the pin is selected.

[image]

StefanSchippers, 2025-03-28:
There are two options to change the selection behavior. The first will switch between incremental select (selected items shown while drawing the selection rectangle) and selection shown at end of drag.

The second item switches the select-by-touch vs select-by-enclosing when dragging to the left.

#### if set show selected elements while dragging the selection rectangle.
#### once selected these can not be unselected by retracting the selection rectangle
#### if not set show selected items at end of drag. Default: enabled (1)
# set incremental_select 0

#### if set select objects if touched by selection rectangle while dragging to the left
#### Default: enabled (1)
# set select_touch 0

Incrementally update selected and unselected objects while extending or retracting the selection rectangle can be done but requires more computation and I don't know if it is worth doing. Nobody asked for that. I need to check the code.

ananthchellappa, 2025-03-28:
Hi Stefan,
      If both these settings are at default (my case), then, is the behavior shown expected? Note that it's not the behaving the same for selection and deselection.. Selection works fine - except for the new thing I noted in the 2nd post..

ananthchellappa, 2025-03-28:
BTW, Stefan, no need to fix. Just reporting.. You are busy.

If you give me a clue as to where in the code to look, I can take a look at the fix..

StefanSchippers, 2025-03-29:
Area selection is started by 
`select_rect(enable_stretch, START,1);` 
(see callback.c:2549)
while moving the selecton is updated by
` select_rect(enable_stretch, RUBBER,1);`
(see callback.c:2519)
on button release the operation is completed by
`select_rect(enable_stretch, END,-1);`
(see callback.c:4464)

closed 2025-05-05 by StefanSchippers