StefanSchippers/xschem #351: Move (click and drag) not working properly if cadence_compat set to 1

ananthchellappa, 2025-04-02:
This one actually shouldn't depend on the variable. 

Do : select all (CTRL-A), then click one on instance and move it.

Observe : without set cadence_compat 1 (i.e, leave it at default, 0), no issue - entire selection moves

If set cadence_compat 1, then only the instance clicked on moves..

[image]

ananthchellappa, 2025-04-02:
Must add, works okay if you use the "m" shortcut (i.e., with cadence_compat = 1 also)

StefanSchippers, 2025-04-02:
This was exactly the request in issue #344
in legacy xschem clicking on an already selected object does nothing.
In cadence_compat you asked to unselect everything and keep only the selected object

ananthchellappa, 2025-04-02:
That is for a click, not click-and-drag.

click : user presses and releases WITHOUT moving the mouse (okay, maybe it moves a few microns)
click and drag : users presses LMB, moves the mouse and then releases LMB.

Come on Stefan :)

closed 2025-05-05 by StefanSchippers