StefanSchippers/xschem #352: "set lvs_netlist 1" uses format-line instead of the lvs_format-line

olisnr, 2025-04-03:
i think this is wrong, but its possible that i do something wrong.

if i try to write an LVS netlist via:

```
	xschem set netlist_type spice
	set lvs_ignore 1
	set lvs_netlist 1
	set spiceprefix 0
	set last_local_netlist_dir $local_netlist_dir
	set local_netlist_dir 0
	xschem netlist [xschem get current_name].cdl
	set local_netlist_dir $last_local_netlist_dir
```

i get the spice-line from after `format` without the spice-prefix. i need the .cdl-line after `lvs_format`...

StefanSchippers, 2025-04-03:
Thank you for reporting. The lvs_netlist variable was thought as a startup (xschemrc) variable, some internal cached data is built based on this variable value. 
I have fixed this.
Now everything is done at runtime, so you can change `lvs_netlist` just before netlisting.

`lvs_netlist` changes netlisting using `lvs_format` rule (if existing) and uncommenting the toplevel .subckt / .ends.

If you **only** want to use `lvs_format` netlisting rule for devices and **nothing else** you better do this:
`xschem set format lvs_format`
to reset to default do this (the empty string will make xschem use the default `format` netlist rule):
`xschem set format {}`

this `xschem set format ...` command allows to put multiple netlist rules for devices and select which one to use.

If you have time, pls update xschem and test again.
Thank you

olisnr, 2025-04-03:
thanks a lot.

i will use `xschem set format lvs_format`

closed 2025-04-03 by olisnr

olisnr, 2025-04-03:
i think the documentation of 
`netlist_type get current netlist type (spice/vhdl/verilog/tedax)` misses lvs

https://xschem.sourceforge.io/stefan/xschem_man/developer_info.html

StefanSchippers, 2025-04-03:
lvs_netlist or any other `xschem set format my_format` you set is a subcategory of spice netlists.
this whole LVS stuff is not used for Verilog / VHDL.

netlist_type tells you if you are using spice or Verilog or VHDL or Tedax ( Tedax is an export  format for pcb, almost unused).

olisnr, 2025-04-03:
ok thanks:

`xschem set netlist_type spice`
`xschem set format lvs_format` or `xschem set format {}`

olisnr, 2025-05-08:
@StefanSchippers now i want to have some bondpads in the LVS netlist but not in the spice-simulation. if i use spice_ignore, then its also missing in the LVS netlist. 
what can i do to get the pad only in the LVS netlis?

StefanSchippers, 2025-05-08:
If you have a bondpad symbol that has only `lvs_format="....."` netlist rule and no` format="...."` rule (or `format=""` empty rule)  this symbol will appear only in lvs netlists. (`xschem set format lvs_format`).
if you set default spice netlist (`xschem set format {}`) bondpads will not be in the netlist.

olisnr, 2025-05-09:
that means that i have to modify the symbol? 
in this case i will use a "dummy-model", because the symbol is from a PDK and to modify it will make problems in the future when downloading a newer version :)

StefanSchippers, 2025-05-09:
You can set the `format` and `lvs_format` in the bondpad instances.
select all placed bondpad instances, press 'q' and add: 
```
format=""
lvs_format="same value as format attribute in the symbol"
```
this will override symbol format attributes.

StefanSchippers, 2025-05-09:
If the main usage of bondpads is to have them **only** in LVS netlist you might consider sending a change request to the pdk maintainers.

olisnr, 2025-05-09:
its from the [iHPs 130nm SiGe BiCMOS](https://github.com/IHP-GmbH/IHP-Open-PDK). they will provide a model in the future i think. at least the capacity against the silicon.