StefanSchippers/xschem #353: Parameterized Schematics (nets, arrays)

itsayellow, 2025-04-03:
What I'm trying to do is make a parameterized circuit that has a number of MOS in series depending on parameter "L":

[image]

I've tried every method to get the "L" to convert to a real value for netlisting and to avoid syntax errors, but nothing works.

Is there a way to do this kind of parameterization with a schematic?  Or is the only way to do this to write a spice description of this circuit and substitute the parameters in that?

StefanSchippers, 2025-04-03:
the label expansion code is already complex enough, adding variable substitution will probably take some time.

For the present you can use tcleval():

[image]

see the backslashes (double) to escape the [ and ] characters that are  special for tcl.
You should create the tcl variable before netlisting.

itsayellow, 2025-04-03:
I'm trying to use parameter "L" from this schematic's symbol.  Is there any way to do that?

itsayellow, 2025-04-03:
So far my only guess at how to do this is to 
1. create a TCL function that makes the entire spice description of the parameterized subcircuit (expanding bus instances itself)
2. In the symbol properties use `format="tcleval(my_custom_function @L )"`

And not use a schematic at all.

StefanSchippers, 2025-04-03:
I think there is some fundamental problem in what you are trying to do. Consider for example a delay symbol that takes as **del** parameter an integer number, may be this is used in the schematic to create a series of **del** unit delay elements (RC or whatever)

[image]

However when you create a spice netlist there is a top level netlist 👍 
```
...
...
x1 IN OUT1 delay del=4
x2 IN OUT2 delay del=8
...
...

```
and the subcircuit netlist, with a default value for the **del** parameter

```
.subckt delay inp outp del=1
...
...
.ends
```

there is one subckt definition for the delay block and spice does not allow to parametrize the netlist structure (like you can do in verilog or VHDL)

you can use the **del** parameter for component values, like:
```
R1 inp outp R='10k * del'
C1 inp GND C='100n * del'
```
but as far as I know you can not create an array of resistors or any other components using the parameter (there is no generate  or similar construct).

One possibility is to use a pcell aka generator, but as you pointed out writing the spice code directly is probably not very different.

StefanSchippers, 2025-04-03:
look at this example that uses a schematic generator that takes 2 parameters: **value** and **n**, and builds a string of `n * r=value` resistors

[test_reschain.zip](https://github.com/user-attachments/files/19592089/test_reschain.zip)

the generator is reschain.tcl.

The trick to quickly write a generator is to draw the circuit for a fixed number of delay elements (say n=4, value=300): 

[image]

then get the .sch file and transform it into a tcl generator (or python if you like):
```
#!/bin/sh
# the next line restarts using wish \
exec tclsh "$0" "$@"
set n [lindex $argv 0]
set i [expr {$n -1}]

set value [lindex $argv 1]

puts "v {xschem version=3.4.6 file_version=1.2}
G {}
K {}
V {}
S {}
E {}
N 300 -190 300 -170 {
lab=P,w\[$i:1\]}
N 300 -110 300 -80 {
lab=w\[$i:1\],M}
C {res.sym} 300 -140 0 0 {name=R\[1:$n\] value=$value}
C {lab_pin.sym} 300 -80 0 1 {name=p2 lab=w\[$i:1\],M}
C {lab_pin.sym} 300 -190 0 1 {name=p3 lab=P,w\[$i:1\]}
C {iopin.sym} 130 -190 0 0 {name=p4 lab=P}
C {iopin.sym} 130 -130 0 0 {name=p6 lab=M}"
```

where the n=4 and value=300 have been parametrized. 

A generator is a program that accepts some arguments and prints on stdout the resulting .sch file.

itsayellow, 2025-04-03:
Ah ok thanks much!  I didn't know there was this mechanism for pcells.  I think this is a cleaner way to do what I was starting to converge to.

itsayellow, 2025-04-03:
I'm trying to do this, but it's not expanding my symbol parameter before it gets to my script.  It shows up as a literal string `"@L"` instead of the value of parameter `L`

StefanSchippers, 2025-04-03:
add a space after @param words. if you do xxxx,tcl(@L) it will try to expand `@L)` instead of `@L`

StefanSchippers, 2025-04-03:
It is a true script. Try to run in a shell:

./reschain.tcl 3 400

The hasbang:
#!/bin/sh
# the next line restarts using wish \
exec tclsh "$0" "$@"

will start as a sh shell script but imediately it exec itself using tclsh, the tcl shell.

the bash shell will "see" the exec line.
The tcl shell will not see the exec line since it is considered as a continuation of the previous comment line.
This is just TCLism... so you don't need to specify the path of tclsh.

itsayellow, 2025-04-03:
> add a space after [@param](https://github.com/param) words. if you do xxxx,tcl([@l](https://github.com/l)) it will try to expand `@L)` instead of `@L`

I was doing that.  But the real problem seems to be originally I used `schematic` as a symbol property inside of my cell.sym.  But apparently it needs it to be an instance property or else it will fail.

itsayellow, 2025-04-03:
Is there a way to instance a pcell schematic and hide the `schematic` parameter on the instance?  It's not something the user should change, it would be nice to have it only inside the symbol.

StefanSchippers, 2025-04-03:
the instance attribute is what is needed to specialize the symbol. Any new combination of n and value generates a new subcircuit.

You can add the schematic attribute in the symbol template. It will be added automatically to the instance. You never need to change it. You only change n and value attributes.
```
type=subcircuit
format="@name @pinlist @symname"
template="name=R1 n=5 value=500 schematic=\"reschain.tcl( @n , @value )\""
default_schematic=ignore
```

georgtree, 2025-04-04:
I use scripts to generate elements with multiple inputs/outputs:

[image]

```tcl
#!/bin/sh
# the next line restarts using wish \
exec tclsh "$0" "$@"

set in_n [lindex $argv 0]
if {$in_n eq {} } { puts stderr "empty arg"; set in_n 2}
if {$in_n < 2} { puts stderr "input number less that 2"; set in_n 2}

for {set i_int 0} {$i_int < $in_n} {incr i_int} {
    lappend in_gain 1.0
    lappend in_offset 0.0
}

set randId [expr {int(rand()*10000)}]

set pinBoxLength 5.0
set pinBoxLengthHlf [expr {$pinBoxLength/2.0}]
set pinLength 10.0
set pinDist 20.0
set boxProj 10.0
set boxWidth 40.0
# coordinate of first pin center
set x0 0.0
set y0 0.0
#define strings
set index 0
set pinStr "@@out0"
set pinStrSubckt "out0"
set pinDefStr "B 5 [expr {$x0-$pinBoxLengthHlf}] [expr {$y0-$pinBoxLengthHlf}] [expr {$x0+$pinBoxLengthHlf}] [expr {$y0+$pinBoxLengthHlf}] {name=out0 dir=out class=elec}\n"
set pLineDefStr "L 4 0 0 -$pinLength 0 {}\n"

for {set i_int 1} {$i_int < $in_n} {incr i_int} {
    # set pin strings
    set pinStr "$pinStr @@out$i_int"
    set pinStrSubckt "$pinStrSubckt out$i_int"
    # shift coordinate of previous pin
    set xShift 0.0
    set yShift $pinDist
    set x0 [expr {$x0+$xShift}]
    set y0 [expr {$y0+$yShift}]
    # calculate output pins coordinates 
    set x0PinBox [expr {$x0-$pinBoxLengthHlf}]
    set y0PinBox [expr {$y0-$pinBoxLengthHlf}]
    set x1PinBox [expr {$x0+$pinBoxLengthHlf}]
    set y1PinBox [expr {$y0+$pinBoxLengthHlf}]    
    set pinDefStr "$pinDefStr\B 5 $x0PinBox $y0PinBox $x1PinBox $y1PinBox {name=out$i_int dir=out class=elec}\n"
    # coordinates of pin line primitive
    set x1PinLine [expr {$x0-$pinLength}]
    set y1PinLine $y0
    set pLineDefStr "$pLineDefStr\L 4 $x0 $y0 $x1PinLine $y1PinLine {}\n"
}
#create body box of symbol
set x0BodyRect $pinLength
set y0BodyRect [expr {-$boxProj} ]
set x1BodyRect [expr {$pinLength+$boxWidth} ]
set y1BodyRect [expr {$pinDist*($in_n-1)+$boxProj} ]
set bodyRect "B 4 -$x0BodyRect $y0BodyRect -$x1BodyRect $y1BodyRect {fill=false}"
puts "V {xschem version=3.4.5 file_version=1.2}
G {}
K {type=xspice_an format=\"@name $pinStr @model \" 
template=\"name=X1 model=dig_source_xspice_id${randId}_@name
device_model=\\\\\".subckt dig_source_xspice_id${randId}_@name $pinStrSubckt
A1 \[$pinStrSubckt\] d_source
.model d_source d_source input_file = \\\\\\\\\\\\\"file.txt\\\\\\\\\\\\\"
.ends dig_source_xspice_id${randId}_@name \\\\\" \" }
S {} 
E {}
$pinDefStr
$pLineDefStr
$bodyRect
T {DSRC} -[expr {($x0BodyRect+$x1BodyRect)/2.0}] [expr {($y0BodyRect+$y1BodyRect)/2.0}] 0 0 0.2 0.2 {layer=7 vcenter=true hcenter=true}
T {@model} -$x1BodyRect [expr {$y0BodyRect-2*$boxProj}] 0 0 0.15 0.15 {hide=true}
T {@name} -$x1BodyRect [expr {$y0BodyRect-1*$boxProj}] 0 0 0.15 0.15 {}

"
```

StefanSchippers, 2025-04-05:
@itsayellow the updated test_reschain example, added also a `test_res_string.sch` for a different example.

The generators are `reschain.tcl` and `res_string.tcl`. The associated symbols are `reschain.sym` and `res_string.sym`.

the reschain.sym symbol has the following attributes:
```
type=subcircuit
format="@name @pinlist @symname"
template="name=XR1 n=5 resvalue=500
+
+ schematic=\"reschain.tcl(@n , @resvalue )\"
+ tclcommand=\"edit_file [abs_sym_path reschain.tcl]\""
extra="schematic tclcommand"
default_schematic=ignore
```

the `extra` attribute tells that schematic and tclcommand are not parameters that should go in the .subckt parameter list unlike `n` and `resvalue`
The + characters are just syntax sugar so when you first instantiate this symbol the instance attributes will be:
```
name=XR3 n=5 resvalue=500
+
+ schematic="reschain.tcl(@n , @resvalue )"
+ tclcommand="edit_file [abs_sym_path reschain.tcl]"
```
the lines beginning with '+' should not be changed.

the res_string.sym is a rc line generator. It takes `n`, `resvalue`, `capvalue` as parameters.

[image]

I have simulated the two test_reschain.sch and test_res_string.sch to verify the netlist is correct.

[test_reschain.zip](https://github.com/user-attachments/files/19614396/test_reschain.zip)

StefanSchippers, 2025-04-07:
Generators are searched  just like sym and sch files. 

[image]

ensure you are putting the tcl extension (if you dont set anything it is included by default).

itsayellow, 2025-04-07:
I think I need to put `schematic=my_library_dir/my_cell.tcl` instead of `schematic=my_cell.tcl`

StefanSchippers, 2025-04-07:
> I don't quite understand what the plusses `+` do. Are they like a comment to the designer, "These are read-only"?

They are completely ignored by xschem. I used them to mark the additional attributes as "inherited by the symbol" and user should not change them. Nothing special.

[image]

StefanSchippers, 2025-04-07:
> I think I need to put `schematic=my_library_dir/my_cell.tcl` instead of `schematic=my_cell.tcl`. Because my library directory is inside the directory in `XSCHEM_LIBRARY_PATH` so that the xschem schematics files don't just have the plain component names, but the include directory above as well.

If you need to put `my_library_dir/my_cell.tcl` you probably have also` my_library_dir/my_cell.sym` as a symbol reference. This means the library one level above `my_library_dir` is in XSCHEM_LIBRARY_PATH.

FYI I also get sometimes a window stuck with no keyboard input, it happened now when editing attributes of a symbol. Still not understood the root cause.

itsayellow, 2025-04-08:
> > I don't quite understand what the plusses `+` do. Are they like a comment to the designer, "These are read-only"?
> 
> They are completely ignored by xschem. I used them to mark the additional attributes as "inherited by the symbol" and user should not change them. Nothing special.

After experimenting I think I understand now.  The properties inside of `template` get dropped directly into the netlist, so if you have more than one line you need the `+`s because it's the line continuation syntax for SPICE.

itsayellow, 2025-05-02:
I think I have the pcells mostly working, but I find now I have a netlist problem, specifically when passing a pin assignment as a parameter.

I'm making a pcell that incudes a pmos3 (parameter specifies body).  I also want the symbol for the parent pcell to accept a parameter for the body and pass it to the internal pmos3.

What happens during netlisting is different for plain schematic vs. pcell.  Schematic (correct): `body` node input to .subckt is also used in pmos3 instances as `body`.  Pcell (error): `body` node input to .subckt is **not** used in pmos3 instances as `body`.  Internal to the pcell subckt definition it uses the dereferenced node name assigned to the `body` parameter of one of the instances of that pcell in the parent schematic.  (e.g. `avdd`). Which results in an internal node with a name that is not connected to the .subckt pin. 

When xschem makes a .subckt for the schematic, it looks like this:
```spice
.subckt pmos3_lx_slvt D G S body  l=2 nfin=1 nf=1 mult=1
*.ipin G
*.iopin D
*.iopin S
XM3 net_int[3] G S body asap7_pmos_slvt nfin=nfin nf=nf m=mult
XM2 net_int[2] G net_int[3] body asap7_pmos_slvt nfin=nfin nf=nf m=mult
XM1 D G net_int[2] body asap7_pmos_slvt nfin=nfin nf=nf m=mult
.ends
```

The details of the circuit are not important, but significant is that it has `body` as an input node to the subckt, and uses `body` on all transistors internally.

When xschem makes a .subckt for the pcell, it looks like this:
```spice
.subckt pmos3_lx_slvt_tcl_1 D G S body  l=2 nfin=1 nf=1 mult=1
*.ipin G
*.iopin D
*.iopin S
XM1 D G S vddin asap7_pmos_slvt nfin=450 nf=1 m=1
.ends
```

The problem: when netlisting the pcell it interprets `body` (in my pcell code output) immediately as `vddin` which is assigned to `body` for one of the pcell instances.  But that doesn't allow the .subckt to properly pass in `body` from the .subckt definition, and I end up with a floating node `vddin` that doesn't connect to anything in the netlist.

I double-quote the `"body"` parameter in my pcell output, and don't use `@body` or anything like that.  Here is my pcell `puts` statement:
```tcl
puts "v {xschem version=3.4.7RC file_version=1.2}
G {}
K {}
V {}
S {}
E {}
N 90 -170 90 -110 {lab=$lab_source}
N 90 -50 90 -20 {lab=$lab_drain}
N -0 -80 50 -80 {lab=G}
N 140 10 170 10 {lab=D}
N 140 -190 170 -190 {lab=S}
C {devices/ipin.sym} 0 -80 0 0 {name=p1 lab=G}
C {devices/iopin.sym} 170 10 0 0 {name=p2 lab=D}
C {devices/iopin.sym} 170 -190 0 0 {name=p3 lab=S}
C {devices/lab_wire.sym} 90 -20 0 0 {name=p4 sig_type=std_logic lab=$lab_drain}
C {devices/lab_wire.sym} 90 -160 0 0 {name=p5 sig_type=std_logic lab=$lab_source}
C {asap7-devices/pmos3_slvt.sym} 70 -80 0 0 {name=$mos_name
nfin=nfin
body=\"body\"
nf=nf
mult=mult
model=pmos_slvt
spiceprefix=X}"
```

Is there a way to prevent xschem from interpreting `body` in the pcell output immediately, and just have it be literal `body`?

StefanSchippers, 2025-05-03:
See this example.
unpack the test_pcell.zip, it will create a test_pcell/ directory. Start xschem from there with:
xschem test_pcell.sch

[test_pcell.zip](https://github.com/user-attachments/files/20022593/test_pcell.zip)

https://github.com/user-attachments/assets/b2940a90-30e1-410a-af89-559af8376139

StefanSchippers, 2025-05-03:
> If I change my `puts` statement inside my .tcl file to have 5 backticks before each quote, it seems to work. (!!!) This seems a little weird to me.

This happens because there are multiple parsing phases on the text file.
The first is the `puts` itself. If you run the generator and see the generated output on stdout you get:

`body=\\"body\\"`

xschem also uses backslash escaping then reading xschem files, 
backslashes are needed for example to allow literal { and } characters that are used in xschem files to group the attribute string:

`C {asap7-devices/pmos3_slvt.sym} 70 -80 0 0 {name=$mos_name body=\\\\\"body\\\\\" mult=mult ...}`

example, use xxx{1} as pmos3 name:

`C {asap7-devices/pmos3_slvt.sym} 70 -80 0 0 {name=xxx\{1\} body=\\\\\"body\\\\\" mult=mult ...}`

Since backslash is used to escape special characters, a literal backslash also needs to be escaped `\\`.
so the assignment:

`body=\\"body\\"`

 becomes:

`body=\"body\"`

the `\"` in xschem means 'get a literal `"` character'

the expression `body="xxx"` will assign `xxx` to body and not `"xxx"`. The double quotes are used to allow entering spaces, like:
`body="xxx yyy"`

StefanSchippers, 2025-05-03:
> The problem: when netlisting the pcell it interprets `body` (in my pcell code output) immediately as `vddin` which is assigned to `body` for one of the pcell instances. But that doesn't allow the [...]

Be careful, what you see in a sub schematic for a parameter (like a W , L or a body node passed via attributes of a transistor) is the resolved value as assigned in parent cells, but if you do a 'q' on it you see the real value. Example below.

[image]

The 'VSS' net label is displayed as VSSA because VSS=VSSA is assigned as attribute in parent symbol instance:

[image]

StefanSchippers, 2025-05-03:
In my opinion you don't need to set a parameter for body in your generator. Just write body=body in your `puts` statement.

The symbol instance will set the actual body value: 

`body=vddin`

and the symbol for your pcell should have global attributes like these:
```
type=subcircuit
format="@spiceprefix@name @@D @@G @@S @body @symname"
template="name=x1 body=VDD"
extra="body"
spiceprefix=X
```

StefanSchippers, 2025-05-03:
I have created a test_xxx.sch testcase that uses a true generator script and sets a body connection.
expand the `test_xxx.zip` into `test_xxx/`, run `xschem test_xxx.sch` from there.

[test_xxx.zip](https://github.com/user-attachments/files/20025384/test_xxx.zip)

https://github.com/user-attachments/assets/b4a4f954-d673-492e-baca-b37ca8324b65

itsayellow, 2025-05-19:
I was away traveling, but am back now.

What you do in the comment above seems to be what I tried, but it's working differently for you than it did for me.  I'll have to see if I can find a difference in what I did.  When I left `body` as plain text, it somehow got substituted in the netlist `.subckt` as one particular instance's value, instead of just keeping the bare word `body`.  If I can find what I did differently I'll report back.

itsayellow, 2025-05-19:
I've already used parameter passing, and even specifying pins via parameters with great success in regular schematics hierarchies.

What I cannot understand is how to make pcells (*.tcl files) work.

Here is my test case.  I tried very hard to replicate your testcase structure, and to get as close to your example syntax as I could.

My problem is that when I include the `schematic=...` in `template` (and thus in the instance) the internal parameters (for example `body`) get overwritten in the netlist subckt by the value from one of my instances.  And I can't find a way to pass the `l` parameter value into the tcl file unless I put that `schematic=` definition inside the `template`.

See the netlist in the zip file.  Instead of `body` in the netlist `.subckt pmos3_lx_slvt_tcl_2` it is de-referenced from one of the instances as `VDD`.  But this won't allow `body` to propagate into the subckt.

itsayellow, 2025-05-19:
I guess a workaround might be to make `@body` another argument to the tcl file.  (??)

But I thought the way I'm trying it should work and was wondering why it doesn't.

StefanSchippers, 2025-05-20:
I have understood the problem.

In some other user cases an instance specifies a model name for the underlying transistors. Ngspice can not set model name as parameters, so the only way to have a subcircuit wilth transistor models set as attributes is to let xschem do the translation.

On the other hand subcircuit pin hierarchies are handles by ngspice so no translation is necessary.

I will create a fix for that.

itsayellow, 2025-05-20:
Thanks, and my apologies for not making a simple test case earlier.  Your testcase zip files helped very much as a framework for mine though!

itsayellow, 2025-05-20:
This fixes the `body` parameter/pin.  Thanks.

However, other parameters I pass to the pcell still get dereferenced in the `.subckt` to the literal values they take for one particular instance of that pcell.  (Instead they should be staying as symbolic parameters.)

Here's a modified test case to display it more clearly:

[test_pmos3lx.zip](https://github.com/user-attachments/files/20353266/test_pmos3lx.zip)

Basically the `pmos3_lx_slvt.tcl` file outputs this for a transistor:
```
C {pmos3_slvt.sym} 70 -80 0 0 {name=$mos_name
nfin=nfin
body=body
nf=nf
mult=mult
model=pmos_slvt
spiceprefix=X
```

However, after netlisting, all those parameters take on literal values instead symbolic names:
```
* expanding   symbol:  pmos3_lx_slvt.tcl( 2 ) # of pins=3
** sym_path: /home/caduser/work-asap7/test_pmos3lx/pmos3_lx_slvt.sym
** sch_path: /home/caduser/work-asap7/test_pmos3lx/pmos3_lx_slvt.tcl
.subckt pmos3_lx_slvt_tcl_2 D G S body  l=2 nfin=1 nf=1 mult=1
*.ipin G
*.iopin D
*.iopin S
XM2 net_int[2] G S body asap7_pmos_slvt nfin=3 nf=1 m=1
XM1 D G net_int[2] body asap7_pmos_slvt nfin=3 nf=1 m=1
.ends
```

In the example, one transistor should have nfin=3 and another nfin=4, but this is not allowed by the hard-coded literal inside the `.subckt`.

StefanSchippers, 2025-05-20:
Ok, this has now been reverted. expressions and params  will be left instead of being calculated from the first instance placed. Can you test again?

itsayellow, 2025-05-21:
I rebuilt with commit ee693aa4 .  However I still see the same netlist behavior as the previous test.  `body` is left symbolic and all other parameters are replaced with literals.

(nfin, nf, m should be left as parameters.)
```
* expanding   symbol:  pmos3_lx_slvt.tcl( 2 ) # of pins=3
** sym_path: /home/caduser/work-asap7/test_pmos3lx/pmos3_lx_slvt.sym
** sch_path: /home/caduser/work-asap7/test_pmos3lx/pmos3_lx_slvt.tcl
.subckt pmos3_lx_slvt_tcl_2 D G S body  l=2 nfin=1 nf=1 mult=1
*.ipin G
*.iopin D
*.iopin S
XM2 net_int[2] G S body asap7_pmos_slvt nfin=3 nf=1 m=1
XM1 D G net_int[2] body asap7_pmos_slvt nfin=3 nf=1 m=1
.ends
```

itsayellow, 2025-05-21:
The workaround setting all parameters as arguments to the tcl file works for me.  I'll be using that in the meantime, but I can still test any commits you want to try using the usual parameterized format.

StefanSchippers, 2025-05-22:
Sorry for the numerous unsuccessful fixes, sometimes a small change that fixes one testcase breaks another.

If you have time can you test again with latest xschem? 

Thank you

itsayellow, 2025-05-22:
No problem!  Thanks for the fixes.

Commit ac629b62 looks like it fixes my testcase pcell netlisting.  This is excellent.  I'm going to test it on a bigger schematic (my real one.)

StefanSchippers, 2025-05-22:
Good!
If you find issues let me know. I am doing small incremental changes to avoid breaking some other test cases.
Thank you for the update.

itsayellow, 2025-05-22:
I just did a test on the full schematic and it also looks great. 

I’m considering it fixed unless I find a problem in the future. Thanks!

closed 2025-06-24 by StefanSchippers