StefanSchippers/xschem #355: How can I move only on vertex?

Dunkelwind, 2025-05-02:
I only want to move this one vertex, how can I do that? Always deleting and redrawing the lines is too faulty. I have tried Shift, Ctrl, Alt variants. No success.

https://github.com/user-attachments/assets/c0cb00ef-87ae-41c0-b67f-3f0997ab12c4

Steffen

StefanSchippers, 2025-05-02:
ctrl-button1 drag **to the right** to select the vertex, then press 'm' for move or simply click over the vertex and move with button down.

https://github.com/user-attachments/assets/29850f33-56c7-451b-afc5-c0aa6a360ac5

StefanSchippers, 2025-05-02:
if you want dragging to the left to behave the same as dragging to the right when selecting objects, set this:
`set select_touch 0`
in your xschemrc. 

By default if dragging to the left touched objects are selected altogether so no vertex-only select is possible.

closed 2025-05-03 by Dunkelwind