StefanSchippers/xschem #356: Label Text Sometimes Put Either On Wire or Text Layer

sifferman, 2025-05-04:
Hello, @StefanSchippers. Thanks for this tool! I am encountering a very weird bug and I wanted to make an issue in case you had an idea of what the problem could be.

In short, recently, running xschem on GitHub actions causes label text to be moved to the symbol text layer. This means that I can't just hide the symbol text.

## Longer Explanation

I am on the latest version of xschem (f0a089fb7102924bc9b2e488274c95f4761ceb98) on Ubuntu 22.04.

I want to display a schematic with label text, but without symbol text. When I generate SVGs on my local computer, the schematic is correct. But on GitHub actions it is incorrect. ([Workflow File](https://github.com/sifferman/sky130_schematics/blob/main/.github/workflows/push.yml))

Both of the following TCL scripts work on my local computer, but do not work on GitHub actions:
```tcl
set enable_layer([xschem get textlayer]) 0
set enable_layer([xschem get pinlayer]) 0
xschem enable_layers
# on GitHub actions, this does nothing
```
```tcl
xschem set sym_txt 0
set enable_layer([xschem get pinlayer]) 0
xschem enable_layers
# on GitHub actions, this disables all text
```

This is the __correct schematic__, which GitHub actions USED to generate, and what my local computer currently generates:

[image]

However, recently, when I generate SVGs through GitHub actions, the label text is somehow being moved to the symbol text layer. So when I disable symbol text, it disables all text:

[image]

Here is the final commit in which the schematics were working: ([release](https://github.com/sifferman/sky130_schematics/releases/tag/2025.04.03.16.45), [job](https://github.com/sifferman/sky130_schematics/actions/runs/14248282194/job/39934660911))

But the schematics became incorrect when I made a documentation-only commit a month later: ([release](https://github.com/sifferman/sky130_schematics/releases/tag/2025.05.02.21.53), [job](https://github.com/sifferman/sky130_schematics/actions/runs/14803948808/job/41568730806)).

When I freeze xschem to an earlier known-working hash ([my change](https://github.com/sifferman/sky130_schematics/commit/dc390e533d30dd745eea13e0db41087bcbd3a318)), the SVG is still incorrect: ([release](https://github.com/sifferman/sky130_schematics/releases/tag/2025.05.03.17.47), [job](https://github.com/sifferman/sky130_schematics/actions/runs/14813193330/job/41590447294)). So I can't do a git-bisect on xschem... Maybe it's a dependency problem? Or maybe there is some non-initialized variable somewhere? Or perhaps a floating pointer?

If you have time, @eliahreeves and I would greatly appreciate any help!

eliahreeves, 2025-05-04:
I am also able to reproduce this using the latest commit compiled on Arch.

sifferman, 2025-05-04:
Also, another small thing is that my local computer compiled xschem with xcb enabled. But on my GitHub actions, it is not enabled. Not sure if that would make a difference.

StefanSchippers, 2025-05-05:
Sorry I don't know what github actions is.

Do you have some problem on your local computer?
Does the rendering problem appear on screen? or in the exported SVG image? or both?
Please send a test schematic and the commands that lead to a wrong result.

I tried all your commands above and the result is consistent, both in the schematic displayed on screen and also in the generated svg.

StefanSchippers, 2025-05-05:
> Also, another small thing is that my local computer compiled xschem with xcb enabled. But on my GitHub actions, it is not enabled. Not sure if that would make a difference.

no xschem built with or without Xcb won't make any difference. if Xcb is not present some functions are done by using the Xlib library. However on almost all Linux systems Xlib is built on top of Xcb so it won't make any difference.

sifferman, 2025-05-05:
This is the script I am running to generate the SVGs. (This is effectively what `make svg/dfxtp/sky130_fd_sc_hd__dfxtp_1.svg` does on <https://github.com/sifferman/sky130_schematics>.)

This script works on my local computer. However, on some computers (like on GitHub's servers) the script generates a schematic without any text. As seen above

```bash
# assume the following schematic is being used:
# https://github.com/sifferman/sky130_schematics/blob/main/schematics/dfxtp/sky130_fd_sc_hd__dfxtp_1.sch

SCHEMATIC=dfxtp/sky130_fd_sc_hd__dfxtp_1 xschem --no_x --log logs/dfxtp/sky130_fd_sc_hd__dfxtp_1.spice.log --script scripts/xschem_generate_svg.tcl
```
```tcl
# https://github.com/sifferman/sky130_schematics/blob/main/scripts/xschem_generate_svg.tcl

set svg_name [xschem get current_dirname]/svg/$::env(SCHEMATIC).svg
set log_file [xschem get current_dirname]/logs/$::env(SCHEMATIC).svg.log

xschem load schematics/$::env(SCHEMATIC).sch

set error_signals {
    "*Symbol not found*"
    "*SKIPPING*"
}
set log_content [read [open $log_file r]]
set found_error 0
foreach signal $error_signals {
    if {[string match $signal $log_content]} {
        puts stderr "Error: $signal"
        set found_error 1
    }
}
if { $found_error } {exit 1}

# Black and White
set dark_colorscheme 0
set light_colors {
    "#ffffff" "#000000" "#000000" "#000000" "#000000" "#000000" "#000000" "#000000"
    "#000000" "#000000" "#000000" "#000000" "#000000" "#000000" "#000000" "#000000"
    "#000000" "#000000" "#000000" "#000000" "#000000" "#000000"}
xschem build_colors

# Enable transparent background for SVG
set transparent_svg 1

# Disable grid lines
set draw_grid 0

# Remove symbol text
# set enable_layer([xschem get textlayer]) 0
xschem set sym_txt 0

# Remove pin layer
set enable_layer([xschem get pinlayer]) 0
xschem enable_layers

xschem redraw

xschem print svg $svg_name
xschem exit closewindow force
```

StefanSchippers, 2025-05-05:
Can I use a shell on the github VM and run some tests?

StefanSchippers, 2025-05-05:
Can you please post here the config.h file that is generated on the GitHub Actions VM after running xschem's configure?

StefanSchippers, 2025-05-05:
I also see you are post processing svg files generated with xschem with inkscape:
```
svg/%.svg: schematics/%.sch
	mkdir -p $(dir $@)
	mkdir -p $(dir logs/$*)
	SCHEMATIC=$* xschem --no_x --log logs/$*.svg.log --script scripts/xschem_generate_svg.tcl
	flock /tmp/inkscape.lock \
	 inkscape $@ --export-overwrite --export-filename=$@ \
	 --actions="select-by-id:rect1;delete;page-fit-to-selection"
```

Can you try to remove this inkscape step?

That said I really don't know how the whole github actions works so I can't really help here.

I have spent half a day learning and googling  all this stuff and i haven't been able to run even a hello-world bash script, not even following many hello world github-actions tutorials. 

Looks like submitting jobs on punched cards on an old IBM-1401 mainframe is easier.

I don't know: 
- where the stuff runs on the VM
-  what are all the ${{...}} variables, where they point to, 
- what is the meaning of most workflow lines, like: 
  - uses: actions/checkout@v4 (why v4? why not v1, or v2 , or v2476457 ) 
  - artifacts: 
  - token:
  - workflow_call:
  -  ... and almost all lines.
  
This piece of workflow:
```
 - name: Generate SVGs
        run: |
          export PDK_ROOT=${{ env.tools }}/volare-pdks
          make all_svg
```
is supposed to generate the svgs using a Makefile.
Where is this Makefile?
is it the one in third_party/ directory? 
so the whole thing is running into that directory? where is this specified?

When running stuff on github actions is there a way to inspect the directories where stuff runs? to see generated logs, files, everything.
A system that can not be inspected in all its single steps is useless.

sifferman, 2025-05-05:
Thanks @StefanSchippers! Your fork worked for me! Now I am just disabling all layers by default, then enabling the ones that I want. ([releases](https://github.com/sifferman/sky130_schematics/releases/tag/2025.05.05.16.54))

[image]

I authored the commit under your name because I was inspired by your solution: <https://github.com/sifferman/sky130_schematics/commit/39e959db265772b4277a3b22d0212768a89bb733>. I also copied over your [signature](<https://github.com/StefanSchippers/sky130_schematics/blob/9465a86a586b0063d22a8b734aedb411d45a9684/docs/CONTRIBUTORS#L9>); hope that's okay. If not, I am happy to revert it.

However, you can see that the pfet "HVT" text is still enabled. This text comes from the symbols: <https://github.com/search?q=repo%3AStefanSchippers%2Fxschem_sky130+%2FT+%7BHVT%7D%2F&type=code>.  Strangely, this text does not show up for me on my local computer, but does show up in the releases.

StefanSchippers, 2025-05-05:
the HVT text is shown because it is on the same layer number 4 that is usually used to draw symbols. Disabling layer 4 will remove all the symbol shapes so it is not a good solution. I think I will do a commit on xschem_sky130 symbols to move it to layer 13, that is already used in FETs to show W/L/mult/nf.
Why does it not show up on your local computer? This is strange. Are you using the same stdcells library?

closed 2025-05-14 by StefanSchippers

sifferman, 2025-05-16:
Thanks! I agree that a solution has been met