StefanSchippers/xschem #362: Stand-alone waveform viewer window

Blebowski, 2025-05-25:
Hello @StefanSchippers,

I recently started using XSCHEM for some of the analog design learning.
The UI is very intuitive when it comes to drawing / netlisting and simulating.
However, the waveform viewer could be improved.

My proposal would be to add a next "openable window" that would contain
the waveform viewer (instead of embedding it in the schematic). The window
would contain:
1. The "graphdialog" as is now displayed when clicking on the embedded waveform plot.
    When loading a RAW file, multiple RAW files could be loaded, each would be displayed
    as foldable object in the "signal list panel". "Signal list panel" would be further hierarchical,
    so a dot separated components of the signal name would form new hierarchy (similar to
    digital waveform viewers).
2. The "waveform plot". The waveform plots would split into "sheets". Each sheet would have
    its settings (like the settings in graphdialog). Sheets would be independent. So one sheet 
    could draw data from a TRAN from one RAW file, next could display data from some type of
    SWEEP. Each sheet would contain "panels". Panels would be displayed below each other. 
    One panel would always be active, that is where the signals would be added. By default, there
    would be only single "panel" that would be active. This is sort-of similar to GAW, but GAW
    UI is really user unfriendly because list of signals is in separate window, and there is no
    zooming with mouse.
3. The zooming could change. To Zoom In/Out on the X axis, "Ctrl" could be used.
    Both GTKWave and Surfer zoom with Ctrl, so it would be nice to keep this consistent accross
    OS waveform viewers. Also a "Shift" could be used to zoom vertically regardless of where the
    cursor is located. The current format where zooming / moving vertically is done only when
    howering over the Y axis is little difficult to grasp.
4. The waveform plot would automatically rescale when rescaling the wrapping window.
5. It would be possible to launch only the waveform window (e.g. by a command line switch).

I will try to create some analog RAW file support in surfer, but my RUST is beginner level,
so it will take me lot of time:
https://gitlab.com/surfer-project/surfer/-/issues/?sort=created_date&state=opened&first_page_size=100&show=eyJpaWQiOiIzOTEiLCJmdWxsX3BhdGgiOiJzdXJmZXItcHJvamVjdC9zdXJmZXIiLCJpZCI6MTY0NzYyNDcyfQ%3D%3D

If you can direct me where to start, I can try to tinker with this myself.

StefanSchippers, 2025-05-25:
These are big changes in waveform viewer. Should be on a separate menu or enabled with a specific configuration, as it breaks completely the current interface many users are now used to.

I added waveform viewing in xschem since I needed a tool to inspect simulation results and at the time no decent open source analog viewer was available, either orphaned or too old or too ugly or too limited or too slow.

In particular I needed the possibility build bit vectors from analog signals:

[image]

One reason for embedding graphs in schematics is that I often place graphs in sub schematics in a hierarchic design. Multiple instances (perhaps at different hierarchy levels) of the same subcircuit will show the related signals. see example in the two amplifiers:

https://github.com/user-attachments/assets/78d0f63f-907c-47e0-a689-8b6b55c7eda7

StefanSchippers, 2025-05-25:
Another reason for embedding graphs in the schematic is live backannotation. You can place / move a (vertical) cursor in the graphs and see voltages and currents annotated in the schematic. Putting graphs on a separate window will make this process more complicated (although it is perfectly possible)

https://github.com/user-attachments/assets/0879c129-15d9-4626-bd4f-bf86a9edab23

rafmag, 2025-05-27:
> Another reason for embedding graphs in the schematic is live backannotation. You can place / move a (vertical) cursor in the graphs and see voltages and currents annotated in the schematic. Putting graphs on a separate window will make this process more complicated (although it is perfectly possible)
> 
>  output.mp4

This is awesome! You should consider at some point making a series of youtube tutorials showing some more advanced features/tricks like this. I've been using xscheme for a while now. and never thought of this!

StefanSchippers, 2025-05-27:
@rafmag : https://www.youtube.com/watch?v=Ka-kAFeaUUs

slagernate, 2025-09-08:
> One reason for embedding graphs in schematics is that I often place graphs in sub schematics in a hierarchic design. Multiple instances (perhaps at different hierarchy levels) of the same subcircuit will show the related signals. see example in the two amplifiers:

This seems very useful, but I can't get this to work easily --would you be able to share a minimum reproducible example (video would be good, a schematic example would be great, but [updated docs](https://xschem.sourceforge.io/stefan/xschem_man/graphs.html) would be best). I may be missing something simple.

StefanSchippers, 2025-09-08:
In general local signals in a subcircuit have the name `<path>.netname`

A net `enable` placed in subcircuit instantiated as `x1` will have name `x1.enable`

the net `enable` placed inside x2, which is placed inside x1, which is a top level instance will have name `x1.x2.enable`

the tcl variable $path expands to the signal path (`x1.x2.` in last example).
This is used in `mos_power_amply.sch` in xschem example `poweramp.sch`:

https://github.com/user-attachments/assets/19a8cc63-2130-4c99-88e8-6ad3cbf4abb6