StefanSchippers/xschem #363: File open pop-up opens behind the "graphview" window

Blebowski, 2025-05-25:
Hi,

when trying to load RAW file from graphdialog, and clicking on "Raw file", the new
window that opens, is behind the graphdialog window. When the graphdialog is
in the middle of the screen, this is really confusing since the newly popped-up
window can't be seen at all.

I am using wayland on ubuntu 22.04.

StefanSchippers, 2025-05-25:
I have added an option to the raw file selector so it will be always on top of graphdialog.
Please test (since I don't see this happening on X11).
Thank you for reporting.

StefanSchippers, 2025-05-26:
I don't have much control over menu display, this is the tk toolkit task.
Can you try to run this:
` wish /usr/share/tcltk/tk8.6/demos/widget`

(may be the demos/widget is installed in a different place on your system (install the `tk8.6-doc` package))

and try one of the menu examples (**Menus and cascades (sub-menus)** ),

check if the blinking happens there too.

closed 2025-05-26 by Blebowski

StefanSchippers, 2025-05-26:
> I see there is a similar problem in:
> [#321](https://github.com/StefanSchippers/xschem/issues/321)
Yes this is surprising. Looks like all emulation layers (XWayland / XQuartz) have very similar issues...