StefanSchippers/xschem #368: Event trigger for new tabs or opened schematics

fawqati-spherical, 2025-07-03:
Hello Stefan,

Ive created a button in the menu thats functional manually, but is dependant on the git lfs locking state of the current schematic. 

I would like its state changed when a new schematic or new tab event happens.

I've looked through the documentation and the examples in the /share folder, but im at a loss.

Your effort and time in advance is sincerely appreciated.

Best, Faisal

StefanSchippers, 2025-07-04:
Hello, Faisal, 
If I understand right you want some user code to be executed when a new schematic is loaded (either in the same window or in another window / tab) Is that correct?

fawqati-spherical, 2025-07-04:
Hi Stefan,

Yes that is correct.  To be clear, I need it to trigger any time the "current active schematic" is changed., either with the open, new tab, decend, ascend etc...

Does Xschem somehow tracks the "current active schematic" or similar ?

FYI, the tcl code i want executed is currently being loaded via an lappend in the xschemrc file.

StefanSchippers, 2025-07-04:
All the actions you mentioned (open, new tab, descend, ascend) have a `load_schematic()` function call in common, so this seems a good place to trigger some user callback code.

fawqati-spherical, 2025-07-04:
> All the actions you mentioned (open, new tab, descend, ascend) have a `load_schematic()` function call in common, so this seems a good place to trigger some user callback code.

I believe that would do the trick. Im assuming load_schematic() is not exposed, but if im wrong, can you point me in the right direction of how i would go about this ?

fawqati-spherical, 2025-07-04:
Thanks. Appreciated

fawqati-spherical, 2025-07-12:
Hello Stefan,
Any update on this issue ?
Best, Faisal

StefanSchippers, 2025-07-12:
A global tcl variable is defined that might contain tcl code. This tcl code is executed any time a new file is loaded:
```
commit f5bd45908243a8e6d440cf71d6a361763b5b14d9 (HEAD -> master, origin/master, origin/HEAD)
Author: stefan schippers <stefan.schippers@gmail.com>
Date:   Wed Jul 9 10:37:18 2025 +0200

    add new xschemrc global variable "load_file_postprocess" that might contain tcl commands to be executed whenever a new file is loaded.
```
If you set this in your xschemrc:
```
set load_file_postprocess {puts [xschem get schname]}
```
the current schematic will be printed any time you open / reload / descend / ascend into a new schematic.

fawqati-spherical, 2025-07-12:
Thanks Stefan. I will update the xschem install, and try this new variable

StefanSchippers, 2025-08-07:
> destroy .menubar.help

In newer xschem releases[ the menu system has ben rewritten](https://github.com/StefanSchippers/xschem/commit/b23b162f2cf0fde9184cd0aca558216bf231731a) to a more modern system. (no need to click to show adjacent menu when moving mouse and more).

The command to delete a menu entry is:
 .menubar delete Help

You can look at [this tcl code](https://github.com/StefanSchippers/xschem/blob/master/src/add_custom_menu.tcl) to see how to create a new menu entry in recent xschem releases.

StefanSchippers, 2025-08-07:
the command to add again the Help menu is:
`.menubar insert { - } cascade -label Help -menu .menubar.help`

The { - } label is the dash that separates the menu entries from the Netlist / Simulate / Waves buttons.
the command inserts the Help entry before the  { - } menu separator entry.

fawqati-spherical, 2025-08-07:
Thanks so much for the info Stefan

Is there any reference you can point me to that would help me with this new modern system of menubar ?

For example, my code below creates a button that i can then dynamically change colour depending on gitlfs locking status of the schematic being editable or not.

button .menubar.sph \
    -pady 0 \
    -highlightthickness 0 \
    -text CHECK \
    -background yellow \
    -foreground black \
    -command {update_sph_button}

pack .menubar.sph -side right

fawqati-spherical, 2025-08-07:
Stefan - i must find a way to thank you !
Please let me know how i can do that

StefanSchippers, 2025-08-07:
No problem, Sorry for the trouble due to the changed menu system, but there were several issues and the old system is likely to be phased out in the future, so I had to change it anyway.

fawqati-spherical, 2025-08-08:
"_A product that doesn't evolve will become obsolete and irrelevant._" - Keep up the great work Stefan !

I have a couple of follow-up questions.

1.  I tried using your set load_file_postprocess {}. It is working but not when i first open xschem. I get the following error :
executing sph_locking_button :
invalid command name "sph_locking_button"

Perhaps the code for sph_locking_button hadnt been loaded yet when i start xschem using the lappend tcl_files ?

2. right now im deleting your Netlist, Simulate and Waves menuitems as we have our own flow and customisations that need to integrate / replace these, but it seems you also have callbacks that are throwing the following errors. Any suggestions how i should do this cleanly ?
```
tclvareval(): error executing .menubar entryconfigure Netlist -background $simulate_bg: bad menu entry index "Netlist"
tclvareval(): error executing .menubar entryconfigure Simulate -background $simulate_bg: bad menu entry index "Simulate"
tclvareval(): error executing .menubar entryconfigure Waves -background $simulate_bg: bad menu entry index "Waves"

```

Ps. feel free to moderate this thread and instruct me to create new issues - if & when you feel they are diverging from the topic of this thread ?

closed 2025-09-21 by StefanSchippers