StefanSchippers/xschem #374: Error from scroll mouse event

trongcuhp, 2025-07-16:
Hi StefanSchippers!
I met the error when I try zoom in and zoom out many times. This cause crash app.

[image]

StefanSchippers, 2025-07-16:
I can not reproduce this issue.

Can you please try with a wired mouse ? (no bluetooth or similar)

If you are using a wireless mouse please also check battery level. Mice that are low on battery send random events that confuse the X server.

What system are you using? Do you have a native X server or Wayland + XWayland?

trongcuhp, 2025-07-17:
I used wired mouse, linux environment and have X server
I haven't Wayland + XWayland
[image]

StefanSchippers, 2025-07-17:
I see you are probably using WSL or WSL2 on Windows?
try to disable 3d acceleration in the WSL environment.

trongcuhp, 2025-07-17:
> I see you are probably using WSL or WSL2 on Windows? If so, try to disable 3d acceleration in the WSL environment. Or have you build xschem with the Cygwin posix layer? in the latter case what is the name of the Xserver package? I recommend using the [Cygwin/X](https://x.cygwin.com/) server, as other Xservers for windows have lot of problems, like Xming, VcXsrv. This is known and it is not a problem I can fix.

Maybe you're right. I'm using these packages in the linux environment:
 apt install cygwin64-gcc-core djgpp-gcc-core gcc-core mingw64-i686-gcc-core mingw64-x86_64-gcc-core -y
apt install flex -y
apt install bison -y
apt install xorg-server xinit xorg-docs xorg-x11-devel -y
apt install tcl-tk-devel -y
apt install libxpm-devel -y
apt install libcairomm1.0-devel -y
apt install libreadline-devel -y
apt install libjpeg-devel -y
apt install libxcb-shape-devel -y
apt install libx11-xcb-devel –y

StefanSchippers, 2025-07-26:
I see, it is again the error 104 (the X server drops connection to the Xschem client).
I have no clue about how to solve this ( I can not reproduce it and I don't know what is going wrong).

trongcuhp, 2025-07-28:
Yes, the Xserver crashed after an error occurred, and I simply restart Xschem to use without to restart the Xserver

StefanSchippers, 2026-09-18:
seems a system problem....

closed 2026-09-18 by StefanSchippers