StefanSchippers/xschem #375: Netlisting: .subckt line dependency on @symname?

kwmartin, 2025-07-20:
I'll try to put an example together when I understand it better; this is just a quick headsup.
1) I updated to most recent version ((3.4.7)
) In my symbol file, if I have a format line as:
`format="@spiceprefix@name @pinlist @symname R=@R m=@m"`
the corresponding .subckt line is as expected:
`.subckt TRN_BEH D G S  R=1K  model=TRN_BEH`
 (almost as expected, I don't understand why "m=1" is not there). However, if I change @symname to @model (this is the only difference - I did the change using an editor to change a single word of the line in the symbol file so the fomat line reads: `format="@spiceprefix@name @pinlist @model R=@R m=@m"
`), the .subckt line is now:
`.subckt TRN_BEH D G S model R m`
which of course is not parsed correctly by ngspice. My guess is that @symname is required in format lines and I couldn't see this mentioned in the documentation (I did a pretty good search but there are many very good changes in the documentation so I may have missed it).
Note: in both cases, the template line is:
`template="name=X1 R=1K m=1 model=TRN_BEH"`

StefanSchippers, 2025-07-21:
Currently there is no specification for a subckt syntax (there is no subckt_format attribute).
The .subckt line in netlists uses the name (without .sym) of the .sym file as the subcircuit name.
I can hopefully make a change to use the `model` attribute (if present) instead of the file name.

in format strings @symname is implicitly taken as the name of the subcircuit and whatever follows @symname is taken as parameter list. If format (for subcircuits) does not contain @symname the parser gets confused.

StefanSchippers, 2025-07-21:
Latest xschem commit allows to use @model as alternative to @symname in "format" attributes.
the "model" attribute must be defined in the symbol template attribute, as you correctly did.

StefanSchippers, 2025-07-21:
> 1. (almost as expected, I don't understand why "m=1" is not there).

the m parameter is special. Every subcircuit instance is multiplied by the m parameter if present in the instance line. This is true for primitive devices:
`R1 pos neg 2k m=3       $ 3 parallel resistors`

as well as for subcircuits:

`X1 plus minus out diff_amp m=3      $ 3 diff_amp circuits in parallel`

subcircuits thus do not need to declare this parameter.

kwmartin, 2025-07-21:
I think all .subckt declarations either have a "params: " or after the subckt name then have "param1=value1" tokens. You could possibly detect the first "param1=value1" and then the token before it is assumed to be the subckt name. Just a thought. Now that I know @symname is required, it shouldn't be a problem.

StefanSchippers, 2025-07-21:
> You could possibly detect the first "param1=value1" and then the token before it is assumed to be the subckt name

Yes, this is a possibility. 
I did not use the '=' recognition to mark the end of the node list in a format string. I used @symname (and now also @model) as a sentinel token. 
I don't remember now, but there was a netlist / spice simulator combination where parameter assignment recognition in format string (to mark the end of node list + subckt name) did not work due to a peculiar simulator syntax.