StefanSchippers/xschem #378: "spiceprefix" not having expected effect.

kwmartin, 2025-07-25:
If my symbol has the following:
`
type=primitive
format="@name @pinlist @VALUE GNOISE=@GNOISE m=@m"
template="name=R1 VALUE=1K GNOISE=1 m=1"
spiceprefix=R
VALUE=@VALUE
`
It netlists fine, but if the format line is changed to:
`
format="@spiceprefix@name @pinlist @VALUE GNOISE=@GNOISE m=@m"
`
All the netlist instantiations start with "X", such as XRL1. I expected them to start with "R" as we have `spiceprefix=R`
I've seen this with many of my "device" symbols. This is not what I expected so there is something I am misunderstanding. Thanks.

StefanSchippers, 2025-07-25:
The symbol spiceprefix is repeated in instances. A single instance may change (override) the symbol spiceprefix attribute.
For example if you have a resistor described with a very complex subcircuit you might want to set spiceprefix=X only sof some instances, while simulating other instances as ordinary resistors, for faster simulation time.

Select all device instances in your design and set spiceprefix=R

StefanSchippers, 2025-07-25:
set attributes of symbol as follows: (spiceprefix=R moved inside template string):

```
type=primitive
format="@spiceprefix@name @pinlist @VALUE GNOISE=@GNOISE m=@m"
template="name=R1 VALUE=1K GNOISE=1 m=1 spiceprefix=R"
```

kwmartin, 2025-07-29:
Additional experiments: if I use something like"
`
format="@prfx@name @pinlist @symname L=@LN1 W=@WN1 NF=@NF AD=@AD AS=@AS PD=@PD PS=@PS"
template="name=N1 VDD=VDD VSS=VSS NF=1 spiceprefix=N prfx=N"
`
and don't have prfx specified in the instance, it does take the template default. If I change the format line to:
`
format="@spiceprefix@name @pinlist @symname L=@LN1 W=@WN1 NF=@NF AD=@AD AS=@AS PD=@PD PS=@PS"
and don't have spiceprefix specified in the instance, it uses "X" as the prefix irrespective of what is specified in the template line. If I specify spiceprefix in the instance, it uses whatever is specified in the instance.

StefanSchippers, 2025-07-29:
I can use any prefix I want and it just works.

Please send a test case I can use here.
Create a zip file with all  *.sch and *.sym files used for the testcase (including sub schematics). 
Include special symbols, do not include standard pdk symbols (like nfet_01v8.sym) or xschem symbols (like vsource.sym, ipin.sym, res.sym).

StefanSchippers, 2025-07-29:
Here the example I have tested. I modified res.sym to have your symbol attributes (right window)
instantiated it into a test schematic (window on the left)

the netlist is correct (lower left window)

So please send the testcase.
Thank you!.

[image]

StefanSchippers, 2025-07-29:
If I set the instance name to C1 it is netlisted as CC1, as expected:

[image]

StefanSchippers, 2025-07-29:
Even if I wrap the above test inside a subcircuit results are as expected:

[image]

kwmartin, 2025-07-30:
I have started putting together a test case, and I am not getting reproducible behavior. Please close this issue, and as soon as I can get a good test case together, I will get back on it. Thank you for your prompt attention.

closed 2025-08-21 by StefanSchippers