StefanSchippers/xschem #387: Copy and paste bug

slagernate, 2025-08-19:
It seems after initiating a copy (with `ctrl-c`) but then cancelling the copy with `esc`, a copy is pasted nonetheless on top of the original copied object(s).

V3.4.7

slagernate, 2025-08-20:
Hi Stefan! Thanks for the quick and detailed response!

This occurs when using: https://github.com/iic-jku/IIC-OSIC-TOOLS/tree/8c1c97bf4df217e54f7f58cbc7003a7589043797

Running `$ cat /headless/.xschem/xschemrc` gives:
```
# Source the PDK xschemrc file
source $env(PDK_ROOT)/$env(PDK)/libs.tech/xschem/xschemrc
```

still occurs with `-i` option/flag.

StefanSchippers, 2025-08-20:
Try to install xschem stand alone and not as part of the docker tools.
see if the issue persist.
https://xschem.sourceforge.io/stefan/xschem_man/install_xschem.html
after doing the test, go in the xschem-src directory do a make uninstall to remove the stand-alone xschem.

slagernate, 2025-08-22:
Will close for now, until I can reproduce with a fresh setup

closed 2025-08-22 by slagernate