StefanSchippers/xschem #392: By default, text box should be selected when editing properties

slagernate, 2025-08-25:
It would be ideal if the text box for editing properties was selected by default when pressing `q` (opening properties window) for quick editing. This is already the case for labels, but would be nice for things like wires, etc. 

Feel free to correct me if this would break current user experience.

StefanSchippers, 2025-08-25:
What are the objects you want text (or part of the text, like a specific attribute) to be selected?
For instances like resistors, capacitors, voltage sources etc the `value` attribute is selected, so by typing you immediately replace its value.

For pins / net labels the `lab` attribute value is selected.

For subcircuit instances I don't know what attribute to highlight since user decides what parameters to pass to the underlying schematic. 
[image]

It is possible to add in the symbols one attribute that tells what to highlight, like `value` or `lab` or `*` to select everything. I like @georgtree suggestion. WIll require some work though.

For text objects I will manage to get the whole text selected. I think this is a nice improvement.

In general I never edit wire attributes directly and I don't recommend doing so.
To set net names use net labels or pins.

Selecting everything is dangerous. User presses one key and _Zapp_... all attribute assignments are gone.

slagernate, 2025-08-26:
Thanks Stefan! Closing for now

closed 2025-08-26 by slagernate