StefanSchippers/xschem #418: `Cancel` button in a waveform graph `graphdialog` does not revert changes made to the graph

slagernate, 2025-09-21:
It is common to make a mistake when editing the signals in the `graphdialog`. But upon hitting `cancel` the changes are not reverted, they are saved no matter what.

Another pain point: if lines of signals in a `graphdialog` text box are shifted (after, e.g., inserting a signal), the color formatting does not move with them. (this is what I am often trying to cancel).

StefanSchippers, 2025-09-22:
the graphdialog immediately applies changes to graph. This is by design. You can even leave it open and double click another graph and see changes as you use the graphdialog.

You can revert all changes done by using the Undo function after closing the graphdialog.

StefanSchippers, 2025-09-22:
> Another pain point: if lines of signals in a `graphdialog` text box are shifted (after, e.g., inserting a signal), the color formatting does not move with them. (this is what I am often trying to cancel).

Yes this needs to be fixed.

StefanSchippers, 2025-09-22:
A Del button has been added to delete a waveform (the one where the cursor is placed in the list of displayed signals). This way of deleting a waveform preserves the colors of others.

slagernate, 2025-09-22:
Really awesome work! And so fast.

Thought about it a bit more and was thinking it would be awesome if, instead of a textbox, signals were displayed / edited with a spreadhseet like view with columns and rows. each row is a signal or bus and columns would delineate signal name, color, signal [dis|en]able etc (rows would still be colored with the signal color). Could even show the hex RGB in the color column (and allow for custom colors without modifying an xschemrc file). Of course this would probably break prior graphs, so just food for thought.

closed 2026-09-17 by slagernate