StefanSchippers/xschem #419: `graphdialog` opens on wrong monitor (when using a 2nd xschem process opened on 2nd monitor)

slagernate, 2025-09-21:


StefanSchippers, 2025-09-22:
I have tested this and I think this is a problem of the window manager.

https://github.com/user-attachments/assets/b6b79040-2d6a-4b2a-bdf2-32022b98c4a6

slagernate, 2026-09-17:
I can reproduce this with a single xschem process on current master: after moving the schematic window to the other monitor, the graph dialog keeps reopening at its old position (and creeps down a bit on each reopen). I've tracked down the cause and would be glad to submit a PR if you're open to it.

StefanSchippers, 2026-09-17:
The dialog remembers size and position when it was last opened. Position is relative to the virtual desktop, so yes, it will reopen in the same (*) position when it was used last.

It is easy to remove the absolute positioning and let the WM decide where to place the dialog box, while maintaining the size / geometry.

(*)
The "creep down a bit on each reopen" is probably a window manager issue. On my system this does not happen, but I know some more modern systems are broken to this aspect.