StefanSchippers/xschem #421: nit: selection area for symbols

slagernate, 2025-09-22:
It would be nice if the selection area for symbols was just the combination of the constituent elements rather than just a rectangle encompassing all them. I've been making many mis-clicks.

slagernate, 2025-09-27:
Relevant: it would also be nice if, when selecting a label, the text itself is a valid to encompass in a selection box (instead of just the anchor point). It's possible to click on a label/text and drag it around, but dragging a box around it does nothing (this feels unintuitive).

slagernate, 2026-09-17:
Happy to submit a PR for either or both of these: (A) symbol click selection based on the symbol's actual drawn elements rather than its bounding box, and (B) net label text counting as part of the label in drag-select. Let me know if you're open to one or both. @StefanSchippers

StefanSchippers, 2026-09-17:
Option (B) is already there. Each component has 2 boundbox coordinates, one including text and one excluding text.

Symbol text in area selects was avoided because it gave so many mis-selections in dense circuit areas.

I can add an option to enable one or the other method.

Option (A) is probably not very simple.

StefanSchippers, 2026-09-18:
Point click of instances has been tuned. Previously distance from mouse to **edges** of bounding box (**without texts**) was used to select the closest instance (given that mouse is inside instance **bbox with texts**)

Now the closest instance is defined as the one with closest distance from mouse to **center** of bbox **with texts**. Such instances are selected only if the mouse is **inside** this bbox.

Please test if you have a chance, let me know if this makes things better or not (less mis-selections).

StefanSchippers, 2026-09-18:
also added a new TCL variable: inst_texts_in_area_select (default setting: 0) 
that you can set in xschemrc to include symbol texts in area selects.