StefanSchippers/xschem #422: verilog + ngpsice cosimulation timing bug: fixed ~2ns offset on outputs

slagernate, 2025-09-22:
Perhaps this is an ngspice bug, but when using e.g. the counter example the output signals appear to be delayed by ~2ns. Becomes apparent when using e.g. `.tran 10p 250n` and ~500p rise/fall times

Tested using Verilator as icarus is still not working for me (even when setting the `timescale) (maybe need to recompile with vvp option or something).

Also tried passing in \`timescale directly ala `ngspice vlnggen -- --timescale 1ps/1ps verilog.v` to no avail.

slagernate, 2025-09-22:
shaved off 1ns of this fixed delay by specifying delay=100ps like so:

```
***Verilator***
device_model=".model my_digital_block d_cosim simulation=\"./my_digital_block.so\" delay=0"
```

slagernate, 2025-09-22:
and shaved off the remaining delay by setting the adc_bridge rise / fall time:

```
device_model=".model comparator adc_bridge in_low={0.9*VDD/2} in_high={1.1*VDD/2} rise_delay=0 fall_delay=0"
```

StefanSchippers, 2025-09-22:
I know the delays are due to the **auto_adc** and **auto_dac** default delays.
I have made all attempts described in the ngspice manual (par. 8.6)  to change these defaults (1ns) with no success.

If you have a working netlist for the counter example please send it to me, so I can update the provided xschem example.

My only last resort is to disable altogether automatic adc/dac bridges and insert manual bridges in the schematic (adc_bridge.sym and dac_bridge.sym)

StefanSchippers, 2025-09-22:
Ok, I figured out the syntax to tune the auto_adc and auto_dac parameters. I have updated the tb_counter_wrapper example so it written in stone and I can copy paste the weird syntaxwhen needed.

```
.control

**** change default parameters of auto adc/dac bridges
pre_set auto_bridge_d_in =
+ ( ".model auto_adc adc_bridge(
+   in_low = '0.9 * 1.8 / 2' in_high = '1.1 * 1.8 / 2'
+   rise_delay=1e-11 fall_delay=1e-11 )"
+ "auto_bridge%d [ %s ] [ %s ] auto_adc" )

pre_set auto_bridge_d_out =
+ ( ".model auto_dac dac_bridge(
+   out_low = 0 out_high = 1.8
+   t_rise=1e-11 t_fall=1e-11 )"
+ "auto_bridge%d [ %s ] [ %s ] auto_dac" )

  save all
  tran 10p 2u
  remzerovec
  write tb_counter_wrapper.raw
.endc
```