StefanSchippers/xschem #423: bug: schematic not marked dirty when written to elsewhere

slagernate, 2025-09-22:
If I have a schematic `aaa.sch` open in two tabs tabA and tabB, with changes made in both, and I write tabB to file, tabA not only will fail to be marked dirty (i.e. different than what is written to file (with an asterisk)), tabA also will not write to file with `ctrl-s`. One must make a slight change to tabA in order to trick xschem into considering it dirty --then `ctrl-s` will work and tabA will be written to file.

This is common when editing `aaa.sch` standalone but also when `xxx.sch` has been opened as a child schematic. My habit is always to hit `ctrl-s` right before closing a schematic, which will write the child `aaa.sch`, overwriting any changes I have made in the original `aaa.sch`.

If `aaa.sch` was marked dirty in all tabs where it is opened, it would help indicate clearly to users that changes have been made to file and the current schematic is dirty.

`aaa.sch` from a tab can simply be marked dirty when that tab is opened (simplifying the fix --it may not be trivial to mark all other duplicate instances of a schematic with every keystroke made).

slagernate, 2025-09-22:
Sorry, I think I misspoke. Let me try again. To reproduce:

1. Create a schematic called `aaa.sch` (lets call this tabA)
2. Add text "AAA" to schematic and save.
3. Open a new tab (lets call this tabB) with `aaa.sch` in it (the warning that pops at this step is great 👍 )
4. Add text "BBB" to this tab (tabB) and save.

At this point, if we click back into tabA, it has a stale design (missing "BBB" text). That's fine, but the real issue is that no asterisk appears above the file.

With the above steps, `ctrl-s` works in tabA to overwrite the new changes in tabB 👍 I am verifying this by `ctrl-s`aving tabA and reloading (`alt-s`) in tabB.

However, surprisingly, if tabB was opened via `e` (from a parent schematic), then it seems writes to `aaa.sch` in tabA do not work (i.e. `alt-s` in tabB does not load the changes from saving tabA).

If this is still unclear I can make a recording to demonstrate. Thanks a ton for your patience and understanding.

slagernate, 2025-09-25:
I see, that makes sense. However, in my humble opinion, if the schematic is identical to that on disk, shouldn't the dirty asterisk go away, similar to e.g. VIM?

StefanSchippers, 2025-09-26:
In edit symbol, now pressing OK with no changes will not mark schematic as modified.

closed 2025-11-10 by slagernate